ros2/ros2_documentation #13: Update for ros2.repos is required

AnandShastry, 2018-11-03:
I tried compiling rosbag2 by pulling latest ros2 ws. rosbag2 uses yaml_cpp_vendor whereas ros2_ws "ros2.repos" doesnt include it. It still has rviz_yaml_cpp_vendor. Can someone please update it

AnandShastry, 2018-11-05:
Actually I am using one given in the installation page...

wget https://raw.githubusercontent.com/ros2/ros2/release-latest/ros2.repos

When I checked in this, I didnt see yaml_cpp_vendor.

dirk-thomas, 2018-11-05:
That url is for the latest release Bouncy. `rosbag` isn't available in Bouncy but targets the upcoming release Crystal. So if you want to build `rosbag` you need to also build the default branches of ROS 2 from source - instead of the latest release.

closed 2018-11-05 by dirk-thomas