ros2/ros2_documentation #67: PyTest dependencies on Ubuntu 18.04

alsora, 2018-12-14:
Hi,
I was following the ros2 crystal install from source instructions.

I'm using a Docker image for Ubuntu 18.04

```
FROM ubuntu:18.04

# working directory
ENV HOME /root
WORKDIR $HOME

# Set timezone otherwise tz will ask for it
ENV TZ=America/Los_Angeles
RUN ln -snf /usr/share/zoneinfo/$TZ /etc/localtime && echo $TZ > /etc/timezone

# general utilities
RUN apt-get update && apt-get install -y \
    curl \
    wget \
    git \
    vim \
    nano \
    python3-pip

# pip
RUN pip3 install --upgrade pip

# Locale options
RUN apt-get install -y locales
```

This is the initial content of the Dockerfile, then I added all the instructions found [here](https://index.ros.org/doc/ros2/Linux-Development-Setup/).
You find the full Dockerfile [here](https://github.com/alsora/ros2-code-examples/blob/master/docker/ros2_crystal_dev/Dockerfile)

Even if I'm on Ubuntu 18.04, I need the following lines
```
RUN python3 -m pip install -U \
  pytest \
  pytest-cov \
  pytest-runner \
  setuptools
```

Otherwise a lots of warnings are prompted about the correct version of pytest not being found and then the build fails while building the `demo_nodes_cpp` package.

This warning appears for several packages

> CMake Warning at /root/ros2_ws/install/ament_cmake_pytest/share/ament_cmake_pytest/cmake/ament_add_pytest_test.cmake:73 (message):
>   The Python module 'pytest' was not found, pytests can not be run (e.g.  on
>   Ubuntu/Debian install the package 'python3-pytest')
> Call Stack (most recent call first):
>   CMakeLists.txt:31 (ament_add_pytest_test)

The error for the `demo_nodes_cpp` package

> CMake Error at CMakeLists.txt:114 (set_property):
>   set_property given TEST names that do not exist:
> 
>     test_tutorial_add_two_ints_server_add_two_ints_client_async__rmw_opensplice_cpp
> 
> Call Stack (most recent call first):
>   /root/ros2_ws/build/demo_nodes_cpp/call_for_each_rmw_implementation.cmake:1 (tests)
>   /root/ros2_ws/install/rmw_implementation_cmake/share/rmw_implementation_cmake/cmake/call_for_each_rmw_implementation.cmake:64 (include)
>   CMakeLists.txt:121 (call_for_each_rmw_implementation)

sloretz, 2018-12-14:
```
# [Ubuntu 16.04] install extra packages not available or recent enough on Xenial
```
That sounds like those packages could be installed from `.deb`s on Bionic, though it looks like CI installs from pip on Bionic too.

```
==> "/home/rosbuild/ci_scripts/venv/bin/python" -m pip install -U pip setuptools
Requirement already up-to-date: pip in /home/rosbuild/ci_scripts/venv/lib/python3.6/site-packages (18.1)
Requirement already up-to-date: setuptools in /home/rosbuild/ci_scripts/venv/lib/python3.6/site-packages (40.6.3)
```

and

```
==> "/home/rosbuild/ci_scripts/venv/bin/python" -m pip install -U EmPy coverage catkin_pkg flake8 flake8-blind-except flake8-builtins flake8-class-newline flake8-comprehensions flake8-deprecated flake8-docstrings flake8-import-order flake8-quotes git+https://github.com/lark-parser/lark.git@0.7b mock nose pep8 pydocstyle pyflakes pyparsing pytest pytest-cov pytest-repeat pytest-rerunfailures pytest-runner pyyaml vcstool colcon-core colcon-defaults colcon-library-path colcon-metadata colcon-output colcon-package-information colcon-package-selection colcon-parallel-executor colcon-powershell colcon-python-setup-py colcon-recursive-crawl colcon-test-result colcon-cmake colcon-ros colcon-bash colcon-zsh
```

I think this could be resolved by a PR to remove the comment.

nuclearsandwich, 2018-12-14:
> That sounds like those packages could be installed from `.deb`s on Bionic, though it looks like CI installs from pip on Bionic too.

This may be either an oversight or a matter of convenient sharing. Perhaps an issue on ros2/ci to suggest using the available versions of these on Bionic is in order (with links back to the documentation updates so we revise the docs if we change)

sloretz, 2018-12-14:
> Perhaps an issue on ros2/ci to suggest using the available versions of these on Bionic is in order (with links back to the documentation updates so we revise the docs if we change)

Created issue ros2/ci#230

closed 2018-12-14 by sloretz