ros2/ros2_documentation #80: Debian Installation Guide Missing important libraries like colcon?

btshrewsbury, 2018-12-18:
I just installed Bouncy on 18.04 using the guide and I was clearly missing important libraries (first obvious one was colcon). I didn't take the time to figure out what was missing, I just ran apt install on all the libraries listed under the build from source section. I believe this either a documentation bug or the debs should rely on and pull in the missing libraries.

marcoesposito1988, 2018-12-27:
The whole binary installation instructions for Crystal seem to be out-of-date with their "from source" counterparts.

In particular, the "pip install" step for macOS is missing colcon-common-extensions (which is not a detail, since ament has been removed)

dirk-thomas, 2018-12-27:
> I just installed Bouncy on 18.04 using the guide ...

@btshrewsbury please include a link to the exact instructions you followed since there are different ones for from source and binary. The development instructions explicitly mention to install `colcon`: https://index.ros.org/doc/ros2/Linux-Development-Setup/#install-development-tools-and-ros-tools

> ... or the debs should rely on and pull in the missing libraries.

Nothing in the Debian packages requires `colcon` in order to run any of the provided executable. `colcon` is a development tool and only needed when you want to build packages from source.

> The whole binary installation instructions for Crystal seem to be out-of-date with their "from source" counterparts.
> In particular, the "pip install" step for macOS is missing colcon-common-extensions (which is not a detail, since ament has been removed)

@marcoesposito1988 The same rational applies to the macOS instructions as for Linux. The instructions to install the binary packages only include the pieces to run the binaries it doesn't setup your environment to actually build packages. That is what the building from source instructions are for.

@ros2/team This confusion seems to be pretty common. I think we should rephase the instructions. A common user even though using the provided binaries wants to build packages too. And just following the binary instructions isn't sufficient for that.

marcoesposito1988, 2018-12-27:
@dirk-thomas 

I see. I am even more confused now. I thought that it would be possible to create an overlay over the binary distribution in order to compile my own packages (just as it was possible to install ROS 1 from apt and then create an overlay to compile my own packages).

If this use case is not supported any more, what is the point of the binary source? Just a demo?

dirk-thomas, 2018-12-27:
> I thought that it would be possible to create an overlay over the binary distribution in order to compile my own packages (just as it was possible to install ROS 1 from apt and then create an overlay to compile my own packages).

That is still possible in the same way. The from source instructions just include tools and dependencies you need in order to build packages (e.g. `build-essential`, `cmake`, `colcon`, and many more) which aren't mentioned in the binary instructions since they are not necessary to run the demos and examples.

What most users want is to use the binary instructions but also install the development tools described in the from source instructions.

marcoesposito1988, 2018-12-27:
Alright, thanks for the fast answer!

Then yes, probably it would be beneficial to add a section to the end of the binary install instructions, where it is pointed out that it is necessary to install colcon (for users coming from ROS 1.. at least I was extremely confused). And maybe a pointer to the ament documentation about overlays, for new users not familiar to catkin.

btshrewsbury, 2018-12-27:
I updated the description to mention it was the debian installation document.  I think most people expect once they install desktop full that they could build and run a hello world example in a ros2 workspace

EwingKang, 2019-01-16:
I second that we need a guide for general user to setup their workspace, and install colcon. Link to the not-so good solution from pull req #95

PR #125 (2019-02-26): Update colcon tutorial
Resolves #80 
Alternative to #95 and #121

Trying to make the tutorial better follow any installation type: binary or source, Linux or OS X or Windows.
Added references to the colcon tutorial at the end of binary installation pages to help direct users who are interested in building packages.

I also reduced the scope of the tutorial to building a single workspace, overlayed on the ROS 2 installation. 
I felt like building a workspace from a repos file was redundant with the installation from source instructions.

---

Admittedly, I overlooked #121 before I did this refa [... cut]
closed 2019-02-26 by jacobperron