ros2/ros2_documentation #202: NVIDIA Jetson Nano - Could not find a package configuration file provided by "test_msgs"

jomoengineer, 2019-05-19:
I'm attempting to following this example with a NVIDIA Jetson Nano and have run into an error that I can not clear. It looks like it is looking for a .cmake file for test_msgs but it can not find one an I was not able to locate it either.
I'm following the install instructions from:
[https://index.ros.org/doc/ros2/Installation/Crystal/Linux-Install-Binary/](https://index.ros.org/doc/ros2/Installation/Crystal/Linux-Install-Binary/)

This is the error from running "colcon build --symlink-install"
<pre><code>
Starting >>> rcl_interfaces
Finished <<< unique_identifier_msgs [13.4s]   
Starting >>> std_msgs
Finished <<< rcl_interfaces [17.6s]                                            
Starting >>> rcl                                           
--- stderr: rcl                                                              
CMake Error at test/CMakeLists.txt:4 (find_package):
  By not providing "Findtest_msgs.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "test_msgs", but CMake did not find one.

  Could not find a package configuration file provided by "test_msgs" with
  any of the following names:

    test_msgsConfig.cmake
    test_msgs-config.cmake

  Add the installation prefix of "test_msgs" to CMAKE_PREFIX_PATH or set
  "test_msgs_DIR" to a directory containing one of the above files.  If
  "test_msgs" provides a separate development package or SDK, be sure it has
  been installed.

---
Failed   <<< rcl	[ Exited with code 1 ]
Aborted  <<< std_msgs                                                    

Summary: 94 packages finished [4min 39s]
  1 package failed: rcl
  1 package aborted: std_msgs
  1 package had stderr output: rcl
  79 packages not processed
</pre></cpde>

jomoengineer, 2019-05-19:
Actually, this is the link I am following.  Perhaps I would better served running from the Debian packages on Ubuntu 18?
[https://index.ros.org/doc/ros2/Installation/Crystal/Linux-Development-Setup/](https://index.ros.org/doc/ros2/Installation/Crystal/Linux-Development-Setup/)

dirk-thomas, 2019-05-20:
According to you routput you are only building a few packages. I assume you have installed the dependencies before. Did you also build/install the package `test_msgs`?

jomoengineer, 2019-05-21:
I only provided the last part of the output from the build where I was following the instructions from the link I provided.

to get past the first test_msgs error, I tried adding the following:

<pre><code>
CMAKE_PREFIX_PATH=$CMAKE_PREFIX_PATH:/home/jetbot/development/ros2_ws/src/ros2/rcl_interfaces/test_msgs
</pre></code>

There is "COLCON_IGNORE" in the test_msgs folder though.

However, when running it again, it failed on the next couple of steps.

Here is the full output of running the colcon command
<pre><code>
$ colcon build --symlink-installStarting >>> ament_package
Starting >>> ament_flake8                      
Finished <<< ament_package [9.43s]                             
Starting >>> ament_cmake_core                 
Finished <<< ament_cmake_core [1.60s]                          
Starting >>> ament_cmake_test
Finished <<< ament_cmake_test [1.42s]                          
Starting >>> ament_cmake_libraries
Finished <<< ament_flake8 [13.3s]                              
Starting >>> ament_cmake_export_libraries
Finished <<< ament_cmake_libraries [1.61s]                                     
Starting >>> ament_cmake_include_directories
Finished <<< ament_cmake_export_libraries [1.63s]
Starting >>> ament_cmake_export_dependencies                           
Finished <<< ament_cmake_include_directories [1.61s]
Starting >>> ament_cmake_export_definitions
Finished <<< ament_cmake_export_dependencies [1.59s]
Starting >>> ament_cmake_export_include_directories
Finished <<< ament_cmake_export_definitions [1.58s]
Starting >>> ament_cmake_export_link_flags
Finished <<< ament_cmake_export_include_directories [1.70s]
Starting >>> ament_cmake_python
Finished <<< ament_cmake_export_link_flags [1.42s]
Starting >>> ament_cmake_export_interfaces
Finished <<< ament_cmake_python [1.40s]                                      
Starting >>> ament_cmake_target_dependencies                            
Finished <<< ament_cmake_export_interfaces [1.54s]
Starting >>> ament_pep257
Finished <<< ament_cmake_target_dependencies [4.90s]
Starting >>> ament_cmake
Finished <<< ament_cmake [1.51s]                               
Starting >>> uncrustify_vendor
Finished <<< uncrustify_vendor [2.40s]                         
Starting >>> ament_cppcheck                    
Finished <<< ament_pep257 [11.9s]                               
Starting >>> ament_copyright
Finished <<< ament_cppcheck [9.19s]                              
Starting >>> ament_uncrustify
Finished <<< ament_copyright [9.94s]                              
Starting >>> ament_lint_cmake
Finished <<< ament_uncrustify [9.93s]                              
Starting >>> ament_cpplint                         
Finished <<< ament_lint_cmake [9.79s]                                  
Starting >>> ament_cmake_lint_cmake
Finished <<< ament_cmake_lint_cmake [1.86s]                         
Starting >>> ament_cmake_copyright                  
Finished <<< ament_cmake_copyright [1.74s]                          
Starting >>> ament_xmllint                          
Finished <<< ament_cpplint [11.8s]                                    
Starting >>> ament_cmake_cppcheck
Finished <<< ament_cmake_cppcheck [1.81s]                            
Starting >>> ament_cmake_cpplint
Finished <<< ament_cmake_cpplint [2.00s]                             
Starting >>> ament_cmake_flake8                      
Finished <<< ament_cmake_flake8 [1.87s]                              
Starting >>> ament_cmake_pep257
[... 145 lines cut ...]
  any of the following names:

    test_msgsConfig.cmake
    test_msgs-config.cmake

  Add the installation prefix of "test_msgs" to CMAKE_PREFIX_PATH or set
  "test_msgs_DIR" to a directory containing one of the above files.  If
  "test_msgs" provides a separate development package or SDK, be sure it has
  been installed.

---
Failed   <<< rcl	[ Exited with code 1 ]
Aborted  <<< rosgraph_msgs                                                  

Summary: 96 packages finished [5min 17s]
  1 package failed: rcl
  1 package aborted: rosgraph_msgs
  1 package had stderr output: rcl
  77 packages not processed
</pre></code>
  

However, following the Linux Install on Debians, I was able to successfully install ROS2 on the Jetson Nano.   
This is the link I with the instructions I followed to get ROS2 installed.
[https://index.ros.org/doc/ros2/Installation/Crystal/Linux-Install-Debians/](https://index.ros.org/doc/ros2/Installation/Crystal/Linux-Install-Debians/)

dirk-thomas, 2019-05-21:
> to get past the first test_msgs error, I tried adding the following:

That is not the right way to to this.

> There is "COLCON_IGNORE" in the test_msgs folder though.

Why?

> following the Linux Install on Debians, I was able to successfully install ROS2 on the Jetson Nano.

So can this ticket be closed? If yes, please go ahead and do so.

jomoengineer, 2019-05-21:
Ah, I think the COLCON_IGNORE was a hold over from trying to install via the ROS2 on arm example.
 [https://github.com/ros2-for-arm/ros2/wiki/ROS2-on-arm-architecture](https://github.com/ros2-for-arm/ros2/wiki/ROS2-on-arm-architecture)

Yeah, I think this can be closed and I will post on the other site if needed.

Thanks.

closed 2019-05-21 by dirk-thomas