ros2/ros2_documentation #209: Python Action Server tutorial is a bit hard to follow

clalancette, 2019-05-22:
I was following the tutorial for the Python Action Server here: https://index.ros.org/doc/ros2/Tutorials/Actions/Writing-an-Action-Server-Python/

One problem that I see with it is that it is a bit hard to follow.  Particularly, it isn't easy to see that each code snippet augments the last one and isn't a complete cut-n-pastable example on its own.  I think it would be better if we restructured it so that either:

1.  Each example was a copy-n-pastable example, or
1.  Each section built on the last somehow (maybe by putting the main in separate file?)

clalancette, 2019-05-22:
The same goes for https://index.ros.org/doc/ros2/Tutorials/Actions/Writing-an-Action-Client-Python/

jacobperron, 2019-05-22:
The intention is to have the additional lines emphasized as we build up the examples. If you build the html locally, you should be able to see this working. Unfortunately, it appears that ROS Index is not styling the emphasized lines appropriately. See https://github.com/ros-infrastructure/rosindex/issues/158.

hidmic, 2019-05-22:
> Each example was a copy-n-pastable example, or

+10

jacobperron, 2019-05-22:
My hope is that the user will type the emphasized lines manually for a better understanding, but at the very least copy-paste those lines into their working example. We could include the entire program at each progression, but I thought it would bloat the tutorial too much.

mm318, 2019-05-22:
https://index.ros.org/doc/ros2/Tutorials/Actions/Writing-an-Action-Client-Python/ mentions `ros2 run action_tutorials fibonacci_action_server.py`, but nowhere is it mentioned how to install `fibonacci_action_server.py` so that it's runnable via `ros2 run` (we only know how to run it by `python3 fibonacci_action_server.py`).

cevans87, 2019-05-22:
In the client tutorial, it is required that someone finished the server tutorial and has the server script available. It is not listed as a prerequisite for the client tutorial.
https://index.ros.org/doc/ros2/Tutorials/Actions/Writing-an-Action-Client-Python/

clalancette, 2019-05-22:
> In the client tutorial, it is required that someone finished the server tutorial and has the server script available. It is not listed as a prerequisite for the client tutorial.

Not quite.  With https://github.com/ros2/demos/pull/343 merged, it should be possible to run the action client tutorial without having gone through the server one.

jacobperron, 2019-05-22:
> https://index.ros.org/doc/ros2/Tutorials/Actions/Writing-an-Action-Client-Python/ mentions ros2 run action_tutorials fibonacci_action_server.py, but nowhere is it mentioned how to install [...]

Good point. It is making the assumption that the package [action_tutorials](https://github.com/ros2/demos/tree/master/action_tutorials) was installed when ROS was installed, which may not be the case.

We can add a step to the prerequisites to ensure that it is installed for use by all the action tutorials.

hidmic, 2019-05-24:
Now that we're releasing an `action_tutorials` package (as of https://github.com/ros2/demos/pull/339), we should be careful about instructing the user to make their own `action_tutorials` package in the tutorial. Sourcing that package before `ros2 run action_tutorials ...` hides available scripts in the underlay (at least on Windows). Either that or rename the package in the tutorial :)

jacobperron, 2019-05-29:
@clalancette Now that the emphasized lines are appearing in the tutorials, PTAL at the tutorials again. 

I'll open up PRs shortly for adding install instructions for `action_tutorials`

jacobperron, 2019-05-29:
See https://github.com/ros2/ros2_documentation/pull/246 regarding the concerns about availability/clashes with the released version of `action_tutorials.

clalancette, 2019-05-29:
> @clalancette Now that the emphasized lines are appearing in the tutorials, PTAL at the tutorials again.

Thanks, that does look better now.  I'll close this out.

closed 2019-05-29 by clalancette