ros2/ros2_documentation #240: ROS1 and ROS2 Bloom instructions contradict each other

brunodmt, 2019-05-28:
The ROS 2 tutorial for [Releasing a ROS 2 package with bloom](https://index.ros.org/doc/ros2/Tutorials/Releasing-a-ROS-2-package-with-bloom/) just says to follow the ROS 1 [Releasing a Package for the First Time](http://wiki.ros.org/bloom/Tutorials/FirstTimeRelease). But at the same time, the step 2 of that tutorial says:

> Note: This procedure will not work for ROS2 packages, since ROS2 uses a new build system. If using ROS2, you will have to create the changelog and release tags yourself. Please see the [ROS 2 Bloom page](https://index.ros.org/doc/ros2/Tutorials/Releasing-a-ROS-2-package-with-bloom/) for more information. 

The problem is that the ROS 2 tutorial doesn't mention anything about changelog and release tags creation.

Is the ROS 2 tutorial missing those steps? Or is the note in the ROS 1 tutorial outdated?

nuclearsandwich, 2019-07-02:
The `catkin_prepare_release` command does not currently update the `setup.py` file of packages built with the ament_python build type. `catkin_generate_changelog` has been working for me in all of the ROS 2 packages I've used it with.

I think the ROS wiki documentation should be updated to note the specific limitation where ament_python packages are concerned or explicitly note that only catkin and ament_cmake packages are currently supported to remain future proof against other build systems.

brunodmt, 2019-07-11:
Makes sense, I wrote this issue while I was reading the guide to use it on an ament_cmake package. In the end I was able to use it without any problem (it was an ament_cmake package).

I'll update the note in the ROS wiki saying that catkin and ament_cmake packages are currently supported.

Thanks!

closed 2019-07-11 by brunodmt

dirk-thomas, 2019-07-24:
> I think the ROS wiki documentation should be updated to note the specific limitation where ament_python packages are concerned or explicitly note that only catkin and ament_cmake packages are currently supported to remain future proof against other build systems.

It would be good to follow up on the documentation side of this.

I updated the `catkin-prepare-release` tool to at least provide correct feedback - that is works for `ament_cmake` packages: ros-infrastructure/catkin_pkg#259