ros2/ros2_documentation #271: Pytest issue on Ubuntu 18.04

borgmanJeremy, 2019-06-03:
My CI started failing a few days ago with the following types of errors (not just for rosidl_adapter).
```
Finished <<< opensplice_cmake_module [4.29s]
stderr: rosidl_adapter‌
CMake Warning at /root/ros2_ws/ros2_base/install/share/ament_cmake_pytest/cmake/ament_add_pytest_test.cmake:73 (message):
  The Python module 'pytest' was not found, pytests can not be run (e.g.  on
  Ubuntu/Debian install the package 'python3-pytest')
Call Stack (most recent call first):
  CMakeLists.txt:31 (ament_add_pytest_test)
```

After some troubleshooting this seems to be due to the fact that when pytest is installed through pip, it installs version 4.6.1 which seems to clobber some of the other ROS python dependencies. 

When this version is installed and I run: python3 -m pytest --version, it fails with a dist error.

If i remove the pip version everything works ok:

```
RUN python3 -m pip uninstall pytest -y
```

I think the build directions should be updated with a workaround for this issue: https://index.ros.org/doc/ros2/Installation/Dashing/Linux-Development-Setup/

dirk-thomas, 2019-06-03:
The referenced instructions explicitly mention to install `pytest` via `pip`. So it is unclear why that didn't work for you.

Uninstalling it isn't the right approach since we rely on a newer version than what is available as a Debian package.

borgmanJeremy, 2019-06-03:
Hrmmm, I'm not sure either. I am building this in a docker container so the environment shouldn't be polluted. Here is a MWE dockerfile that follows copies and pastes from the directions and outputs the pytest error. 

```

FROM ubuntu:18.04

RUN echo 'Etc/UTC' > /etc/timezone && \
  ln -s /usr/share/zoneinfo/Etc/UTC /etc/localtime && \
  apt-get update && apt-get install -q -y tzdata curl gnupg2 lsb-release \
  && curl http://repo.ros2.org/repos.key | apt-key add - \
  && sh -c 'echo "deb [arch=amd64,arm64] http://packages.ros.org/ros2/ubuntu `lsb_release -cs` main" > /etc/apt/sources.list.d/ros2-latest.list'  \
  && rm -rf /var/lib/apt/lists/*

RUN apt-get update && apt install -y \
  build-essential \
  cmake \
  git \
  python3-colcon-common-extensions \
  python3-lark-parser \
  python3-lxml \
  python3-numpy \
  python3-pip \
  python-rosdep \
  python3-vcstool \
  wget \
# install some pip packages needed for testing
 && python3 -m pip install -U \
  argcomplete \
  flake8 \
  flake8-blind-except \
  flake8-builtins \
  flake8-class-newline \
  flake8-comprehensions \
  flake8-deprecated \
  flake8-docstrings \
  flake8-import-order \
  flake8-quotes \
  pytest-repeat \
  pytest-rerunfailures \
  pytest \
  pytest-cov \
  pytest-runner \
  setuptools\
# install Fast-RTPS dependencies
 && apt install --no-install-recommends -y \
  libasio-dev \
  libtinyxml2-dev

RUN python3 -m pytest --version

```

Error
```
Step 6/6 : RUN python3 -m pytest --version
 ---> Running in 4c2ae13d788f
Traceback (most recent call last):
  File "/usr/lib/python3.6/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/lib/python3.6/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/usr/local/lib/python3.6/dist-packages/pytest.py", line 102, in <module>
    raise SystemExit(pytest.main())
  File "/usr/local/lib/python3.6/dist-packages/_pytest/config/__init__.py", line 63, in main
    config = _prepareconfig(args, plugins)
  File "/usr/local/lib/python3.6/dist-packages/_pytest/config/__init__.py", line 207, in _prepareconfig
    pluginmanager=pluginmanager, args=args
  File "/usr/local/lib/python3.6/dist-packages/pluggy/hooks.py", line 289, in __call__
    return self._hookexec(self, self.get_hookimpls(), kwargs)
  File "/usr/local/lib/python3.6/dist-packages/pluggy/manager.py", line 87, in _hookexec
    return self._inner_hookexec(hook, methods, kwargs)
  File "/usr/local/lib/python3.6/dist-packages/pluggy/manager.py", line 81, in <lambda>
    firstresult=hook.spec.opts.get("firstresult") if hook.spec else False,
  File "/usr/local/lib/python3.6/dist-packages/pluggy/callers.py", line 203, in _multicall
    gen.send(outcome)
  File "/usr/local/lib/python3.6/dist-packages/_pytest/helpconfig.py", line 94, in pytest_cmdline_parse
    config = outcome.get_result()
  File "/usr/local/lib/python3.6/dist-packages/pluggy/callers.py", line 80, in get_result
    raise ex[1].with_traceback(ex[2])
  File "/usr/local/lib/python3.6/dist-packages/pluggy/callers.py", line 187, in _multicall
    res = hook_impl.function(*args)
  File "/usr/local/lib/python3.6/dist-packages/_pytest/config/__init__.py", line 687, in pytest_cmdline_parse
    self.parse(args)
  File "/usr/local/lib/python3.6/dist-packages/_pytest/config/__init__.py", line 895, in parse
    self._preparse(args, addopts=addopts)
  File "/usr/local/lib/python3.6/dist-packages/_pytest/config/__init__.py", line 836, in _preparse
    self._consider_importhook(args)
  File "/usr/local/lib/python3.6/dist-packages/_pytest/config/__init__.py", line 784, in _consider_importhook
    self._mark_plugins_for_rewrite(hook)
  File "/usr/local/lib/python3.6/dist-packages/_pytest/config/__init__.py", line 806, in _mark_plugins_for_rewrite
    for name in _iter_rewritable_modules(package_files):
  File "/usr/local/lib/python3.6/dist-packages/_pytest/config/__init__.py", line 625, in _iter_rewritable_modules
    for fn in package_files:
  File "/usr/local/lib/python3.6/dist-packages/_pytest/config/__init__.py", line 803, in <genexpr>
    for file in dist.files
TypeError: 'NoneType' object is not iterable
The command '/bin/sh -c python3 -m pytest --version' returned a non-zero code: 1

```

pbaughman, 2019-06-04:
I'm seeing the same issue in my CI job that runs on the nightly docker image.  Things passed on May 31st with docker image `sha256:1d4982c31809869c5b59892e9eaf014cb149acef28342fe6c0832dc3c586a964 for osrf/ros2:nightly`

Then the same job started dying on June 2nd with docker image `sha256:b364911fb43ef375043154ba0d72a3787983a0941cb55ccc0fd589b1d7976135 for osrf/ros2:nightly`

I can reproduce the issue locally.  I don't have the 1d4982 docker image available, but I can test with one from a week ago and it works fine.

My CI pipelines [are here](https://gitlab.com/ApexAI/apex_rostest/pipelines).  I will continue to investigate

pbaughman, 2019-06-04:
Sorry forgot to include this above. simplest repro case I have right now is:

```
$ docker pull osrf/ros2:nightly
$ docker run --rm -it osrf/ros2:nightly
[connext_cmake_module] Warning: The location at which Connext was found when the workspace was built [[/opt/rti.com/rti_connext_dds-5.3.1]] does not point to a valid directory, and the NDDSHOME environment variable has not been set. Support for Connext will not be available.
root@c1a178e62401:/# pytest
Traceback (most recent call last):
  File "/usr/local/bin/pytest", line 11, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.6/dist-packages/_pytest/config/__init__.py", line 63, in main
    config = _prepareconfig(args, plugins)
  File "/usr/local/lib/python3.6/dist-packages/_pytest/config/__init__.py", line 207, in _prepareconfig
    pluginmanager=pluginmanager, args=args
  File "/usr/local/lib/python3.6/dist-packages/pluggy/hooks.py", line 289, in __call__
    return self._hookexec(self, self.get_hookimpls(), kwargs)
  File "/usr/local/lib/python3.6/dist-packages/pluggy/manager.py", line 87, in _hookexec
    return self._inner_hookexec(hook, methods, kwargs)
  File "/usr/local/lib/python3.6/dist-packages/pluggy/manager.py", line 81, in <lambda>
    firstresult=hook.spec.opts.get("firstresult") if hook.spec else False,
  File "/usr/local/lib/python3.6/dist-packages/pluggy/callers.py", line 203, in _multicall
    gen.send(outcome)
  File "/usr/local/lib/python3.6/dist-packages/_pytest/helpconfig.py", line 94, in pytest_cmdline_parse
    config = outcome.get_result()
  File "/usr/local/lib/python3.6/dist-packages/pluggy/callers.py", line 80, in get_result
    raise ex[1].with_traceback(ex[2])
  File "/usr/local/lib/python3.6/dist-packages/pluggy/callers.py", line 187, in _multicall
    res = hook_impl.function(*args)
  File "/usr/local/lib/python3.6/dist-packages/_pytest/config/__init__.py", line 687, in pytest_cmdline_parse
    self.parse(args)
  File "/usr/local/lib/python3.6/dist-packages/_pytest/config/__init__.py", line 895, in parse
    self._preparse(args, addopts=addopts)
  File "/usr/local/lib/python3.6/dist-packages/_pytest/config/__init__.py", line 836, in _preparse
    self._consider_importhook(args)
  File "/usr/local/lib/python3.6/dist-packages/_pytest/config/__init__.py", line 784, in _consider_importhook
    self._mark_plugins_for_rewrite(hook)
  File "/usr/local/lib/python3.6/dist-packages/_pytest/config/__init__.py", line 806, in _mark_plugins_for_rewrite
    for name in _iter_rewritable_modules(package_files):
  File "/usr/local/lib/python3.6/dist-packages/_pytest/config/__init__.py", line 625, in _iter_rewritable_modules
    for fn in package_files:
  File "/usr/local/lib/python3.6/dist-packages/_pytest/config/__init__.py", line 803, in <genexpr>
    for file in dist.files
TypeError: 'NoneType' object is not iterable
```

dirk-thomas, 2019-06-04:
This looks like a regression in `pytest` 4.6.0 (released Jun 1st) as well as 4.6.1 and 4.6.2. The problem doesn't happen in 4.5 and older.

The bug appears when the `files` attribute of a `importlib_metadata.PathDistribution` is `None`. See pytest-dev/pytest#5389 for a proposed fix.

borgmanJeremy, 2019-08-28:
Sorry should have closed it, its been resolved.