ros2/ros2_documentation #277: Add C++ Action tutorials

jacobperron, 2019-06-07:
Follow-up from #166. We currently have Python tutorials, but there are only place-holder pages for the C++ tutorials.

Similar to the existing Python tutorials, the code can live in the [action_tutorials](https://github.com/ros2/demos/tree/master/action_tutorials) package. To avoid maintaining multiple copies of the tutorial code, we might consider moving the entire tutorial into the action_tutorials package and reference it from the ROS 2 documentation.

jacobperron, 2020-03-17:
@maryaB-osr Let me know what you think about the existing [Action tutorials](https://index.ros.org/doc/ros2/Tutorials/Actions/) and if you have plans for refactoring. I think it would be nice to do the C++ tutorials (which I can do) before Foxy.

maryaB-osr, 2020-03-17:
@jacobperron I do have an idea for refactoring, let me put that together and then you can fill in the C++ code based on the new structure

clalancette, 2020-12-21:
This was done in #902, so closing this out.

closed 2020-12-21 by clalancette