ros2/ros2_documentation #329: Update troubleshooting page with more info and add to install page.

piraka9011, 2019-09-16:
While the current troubleshooting page does get people on the right track, it does not directly provide solutions to addressing certain issues such as multicasting. Moreover, it is not really visible to new users (aside from the sidebar...).

My suggestion is to:

1. Add the [pub/sub demo](https://index.ros.org/doc/ros2/Tutorials/Colcon-Tutorial/#id15) to the [Debian installation instructions page](https://index.ros.org/doc/ros2/Installation/Dashing/Linux-Install-Debians/) (since they are only on the building from source instructions).
2. Add a link to [Troubleshooting](https://index.ros.org/doc/ros2/Troubleshooting/) on the Debian install instructions page.
3. Update the Troubleshooting page with instructions on how to fix the multicasting issue:
    1. Check `ifconfig` and see if the `MULTICAST` flag is the for `lo`:
    2. If not, allow UDP multicast in the ufw
       ```bash
       sudo ufw allow in proto udp to 224.0.0.0/4
       sudo ufw allow in proto udp from 224.0.0.0/4
       ```

@maryaB-osr @wjwwood if this sounds good, I will open a PR for this.

maryaB-osr, 2019-09-16:
1. Agree on the pub/sub demo. I don't think any of the other installation pages are missing it, but if they are, it should be added there too. 

2. The Debian installation doesn't have it's own troubleshooting section, so it makes sense for that guide to have a link to the troubleshooting page at the end. I think the best thing would be for _all_ the troubleshooting sections across all the installation guides to be consolidated on that main troubleshooting page. And then the end of each installation guide can link to, not only the troubleshooting page, but the exact section of the page that corresponds to that particular installation. The idea being to clean up the installation guides. 

I'm not saying you have to do all that @piraka9011 :), I'm just letting you know what the end goal is so you can act accordingly.

piraka9011, 2019-11-07:
Closing due to #396

closed 2019-11-07 by piraka9011