ros2/ros2_documentation #352: Migration guide for parameters

jacobperron, 2019-10-09:
I'm not aware of any existing resource about the changes for parameters going from ROS 1 to ROS 2. In particular, I think it would be useful if there was at least a brief about how to update existing parameter YAML files to be ROS 2 compatible. Perhaps as part of the [Migration Guide](https://index.ros.org/doc/ros2/Contributing/Migration-Guide/). I don't know the details of the necessary changes, but I believe there are some.

Blast545, 2020-08-24:
I can help closing this issue. I read the design document for the change between the past Parameter Server vs the new parameters assigned to nodes functionality and there is a brief section explaining how to implement a similar feature in ROS 2. https://design.ros2.org/articles/ros_parameters.html

What could be the best example scenario for the Migration Guide? Maybe it will be needed an additional document with the command line differences for "ros2 param" vs "ros param", WDYT? @jacobperron

If there was needed an additional document to outline the differences, where do you think it should go? @maryaB-osr

jacobperron, 2020-08-25:
@Blast545 Great!

Here are some things that I think are worth adding to the migration guide:

- Summarizing the difference between the ROS 1 parameter server model and how parameters are tied to nodes in ROS 2
  - we can add a reference to the design doc for more details.
- Differences between a ROS 1 parameter YAML file and the ROS 2 parameter YAML file.
  - E.g. there are some small changes users should add like the node name(s) and `ros__parameters` key.
- Briefly mention some new features, like passing parameters on the command-line (ie. individual parameters or a parameters file)

I don't think we should explain how to re-create the parameter server from ROS 1.
Also, I don't think we need to describe differences between the command line tools (ros2 param vs ros param); there's a whole [tutorial for that](https://index.ros.org/doc/ros2/Tutorials/Parameters/Understanding-ROS2-Parameters/), which we can simply reference in the migration guide.

In the Migration Guide, there is a [reference to a separate migration page for launch files](https://index.ros.org/doc/ros2/Contributing/Migration-Guide/#launch-files). I think we could do something similar for parameters; create a separate article for migrating ROS 1 parameters to ROS 2.

jacobperron, 2020-10-13:
Done in https://github.com/ros2/ros2_documentation/pull/887

closed 2020-10-13 by jacobperron