ros2/ros2_documentation #381: New tutorial about rosdep

clalancette, 2019-10-24:
In https://github.com/ros2/ros2_documentation/pull/359#discussion_r335643570 , we discussed the fact that it would be nice to have a ROS 2 tutorial talking about rosdep in more detail.  This issue tracks the addition of that tutorial to ros2_documentation.

It should probably talk about:
* What rosdep is
* It's relationship to the package.xml
* The kinds of keys that rosdep will work with (ROS distribution keys, package keys, pip keys)
* How to use rosdep on the command-line
* How to submit new keys to rosdistro (or at least point to other documentation that describes that)

dirk-thomas, 2019-10-24:
It sounds to me that a lot of that information should be available somewhere. It is probably better to reference that than coming up with redundant information.

jacobperron, 2019-10-24:
> It is probably better to reference that than coming up with redundant information.

Or at least put somewhere where it can be referenced by ROS 1 and ROS 2 sources.

maryaB-osr, 2019-10-25:
> It is probably better to reference that than coming up with redundant information.

To be fair, there is a [ROS 1 tutorial on rosdep](http://wiki.ros.org/ROS/Tutorials/rosdep), and we are porting the tutorials. 

Has anything changed between ROS 1 and ROS 2 regarding rosdep? I couldn't find any ROS2-centric documentation on it. I'd prefer to avoid linking to the ROS wiki as much as possible (unless that's something that's already done a lot on rosindex? not sure).

SteveMacenski, 2022-05-13:
[image]

(its Friday, we can have fun on Fridays, right?)

SteveMacenski, 2022-05-13:
https://github.com/ros2/ros2_documentation/pull/2561

gavanderhoorn, 2022-07-06:
Seems like @SteveMacenski addressed this.

SteveMacenski, 2022-07-06:
Agreed, I don't have permissions to close this however

clalancette, 2022-07-06:
Yep, agreed.  Closing.

closed 2022-07-06 by clalancette