ros2/ros2_documentation #454: ubuntu 16.04 can't install ros-ardent-turtlesim and ros-ardent-rqt*

cuijianaaa, 2019-12-02:
I have followed https://index.ros.org/doc/ros2/Installation/Crystal/Linux-Install-Debians/ to install ros2, but can't install rqt and turtlesim, anything have I missed? 
~$ sudo apt install ros-ardent-turtlesim
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package ros-ardent-turtlesim

maryaB-osr, 2019-12-02:
I believe ROS 2 turtlesim only exists for [Dashing](https://index.ros.org/doc/ros2/Installation/Dashing/Linux-Install-Debians/) and [Eloquent](https://index.ros.org/doc/ros2/Installation/Eloquent/Linux-Install-Debians/). I would suggest downloading one of those, favorably Eloquent since it's the latest.
Also, if you're [trying the tutorials](https://index.ros.org/doc/ros2/Tutorials/#beginner-level), they're meant for Eloquent.

cuijianaaa, 2019-12-02:
@maryaB-osr Thanks for reply so fast.

PR #455 (2019-12-02): Clarify tutorial prereqs for installing turtlesim/rqt
Adding a couple sentences about what needs to be in place to use `sudo apt install ros-<distro>-turtlesim/rqt*`

Fixes #454 (more like "answers", since it's a question)
closed 2019-12-02 by maryaB-osr