ros2/ros2_documentation #461: "try the examples" doesnt work....

fishworx, 2019-12-07:
see link below... examples didnt work...

http://brisbaneroboticsclub.id.au/install-ros2-foxy-fitzroy-on-nvidia-nano/

dirk-thomas, 2019-12-07:
Please include enough information what you did. Also don't reference that from externally but post it here in the issue. And don't use images where text is sufficient since it is not searchable.

From a brief look to the screenshot your workspace seems to not contain all packages. So something before which isn't shown didn't follow the tutorial.

fishworx, 2019-12-08:
marty@marty-desktop:~$ . /ros2_eloquent/install/local_setup.bash
marty@marty-desktop:~$ ros2 run demo_nodes_cpp talker
bash: ros2: command not found
marty@marty-desktop:~$ cd ros2_eloquent/
marty@marty-desktop:~/ros2_eloquent$ cd install/
marty@marty-desktop:~/ros2_eloquent/install$ ros2 run demo_nodes_cpp talker
bash: ros2: command not found
marty@marty-desktop:~/ros2_eloquent/install$ ^C
marty@marty-desktop:~/ros2_eloquent/install$ 

thanks Dirk, the error msg is above...

I followed these instructions, there were no install errors.

https://index.ros.org/doc/ros2/Installation/Eloquent/Linux-Development-Setup/

I'm a bit of a Ron Burgandy when it comes to installs from linux CLI, I just enter the commands from the instructions...

May I make one recommendation that I've found useful.... after entering a command, test the command worked... i.e. type 'ls' and ensure the files are in the directory....

Is there a way top test which packages should be installed etc??

dirk-thomas, 2019-12-08:
> the error msg is above...

Unfortunately that is not the first error message. Something else must have failed before since your `install` directory only contains a few packages and e.g. lacks the one providing the `ros2` commands.

So please include more information of each and every step you did including the output. You probably want to go through the instruction from the beginning one more time.

fishworx, 2019-12-09:
tried reinstall again... same error.... is there someway to check which install command failed?

dirk-thomas, 2019-12-09:
If you don't see any problems maybe when posting the whole sequence here someone can spot something.

fishworx, 2019-12-09:
how do i post a whole sequence?

dirk-thomas, 2019-12-09:
You mark the commands and output in the terminal e.g. using your mouse and paste them here as a comment (enclosed in triple backticks).

fishworx, 2019-12-09:
set locale - okay

'''marty@marty-desktop:~$ sudo locale-gen en_US en_US.UTF-8
[sudo] password for marty: 
Generating locales (this might take a while)...
  en_US.ISO-8859-1... done
  en_US.UTF-8... done
Generation complete.
marty@marty-desktop:~$ sudo update-locale LC_ALL=en_US.UTF-8 LANG=en_US.UTF-8
marty@marty-desktop:~$ export LANG=en_US.UTF-8
'''

fishworx, 2019-12-09:
add ros 2 repository - okay

'''
marty@marty-desktop:~$ sudo apt update && sudo apt install curl gnupg2 lsb-release
Get:1 file:/var/cuda-repo-10-0-local-10.0.326  InRelease
Ign:1 file:/var/cuda-repo-10-0-local-10.0.326  InRelease
Get:2 file:/var/visionworks-repo  InRelease
Ign:2 file:/var/visionworks-repo  InRelease
Get:3 file:/var/visionworks-sfm-repo  InRelease
Ign:3 file:/var/visionworks-sfm-repo  InRelease
Get:4 file:/var/visionworks-tracking-repo  InRelease
Ign:4 file:/var/visionworks-tracking-repo  InRelease
Get:5 file:/var/cuda-repo-10-0-local-10.0.326  Release [574 B]
Get:5 file:/var/cuda-repo-10-0-local-10.0.326  Release [574 B]
Get:6 file:/var/visionworks-repo  Release [1,999 B]
Get:6 file:/var/visionworks-repo  Release [1,999 B]                            
Get:7 file:/var/visionworks-sfm-repo  Release [2,003 B]                        
Get:7 file:/var/visionworks-sfm-repo  Release [2,003 B]                        
Get:8 file:/var/visionworks-tracking-repo  Release [2,008 B]                   
Get:8 file:/var/visionworks-tracking-repo  Release [2,008 B]                   
Hit:9 http://international.download.nvidia.com/jetson/repos/common r32 InRelease
Hit:10 http://international.download.nvidia.com/jetson/repos/t210 r32 InRelease
Hit:15 http://packages.ros.org/ros2/ubuntu bionic InRelease                    
Hit:16 http://ports.ubuntu.com/ubuntu-ports bionic InRelease             
Get:17 http://ports.ubuntu.com/ubuntu-ports bionic-updates InRelease [88.7 kB]
Get:18 http://ports.ubuntu.com/ubuntu-ports bionic-backports InRelease [74.6 kB]
Get:19 http://ports.ubuntu.com/ubuntu-ports bionic-security InRelease [88.7 kB]
Get:20 http://ports.ubuntu.com/ubuntu-ports bionic-updates/main arm64 DEP-11 Metadata [290 kB]
Get:21 http://ports.ubuntu.com/ubuntu-ports bionic-updates/main DEP-11 48x48 Icons [73.8 kB]
Get:22 http://ports.ubuntu.com/ubuntu-ports bionic-updates/main DEP-11 64x64 Icons [143 kB]
Get:23 http://ports.ubuntu.com/ubuntu-ports bionic-updates/universe arm64 DEP-11 Metadata [259 kB]
Get:24 http://ports.ubuntu.com/ubuntu-ports bionic-updates/universe DEP-11 48x48 Icons [197 kB]
Get:25 http://ports.ubuntu.com/ubuntu-ports bionic-updates/universe DEP-11 64x64 Icons [445 kB]
Get:26 http://ports.ubuntu.com/ubuntu-ports bionic-backports/universe arm64 DEP-11 Metadata [7,976 B]
Get:27 http://ports.ubuntu.com/ubuntu-ports bionic-security/main arm64 DEP-11 Metadata [32.5 kB]
Get:28 http://ports.ubuntu.com/ubuntu-ports bionic-security/main DEP-11 48x48 Icons [17.6 kB]
Get:29 http://ports.ubuntu.com/ubuntu-ports bionic-security/main DEP-11 64x64 Icons [41.5 kB]
Get:30 http://ports.ubuntu.com/ubuntu-ports bionic-security/universe arm64 DEP-11 Metadata [37.0 kB]
Get:31 http://ports.ubuntu.com/ubuntu-ports bionic-security/universe DEP-11 48x48 Icons [16.4 kB]
Get:32 http://ports.ubuntu.com/ubuntu-ports bionic-security/universe DEP-11 64x64 Icons [111 kB]
Fetched 1,924 kB in 7s (292 kB/s)                                              
Reading package lists... Done
Building dependency tree       
Reading state information... Done
All packages are up to date.
Reading package lists... Done
Building dependency tree       
Reading state information... Done
lsb-release is already the newest version (9.20170808ubuntu1).
curl is already the newest version (7.58.0-2ubuntu3.8).
gnupg2 is already the newest version (2.2.4-1ubuntu1.2).
The following packages were automatically installed and are no longer required:
  apt-clone archdetect-deb bogl-bterm busybox-static cryptsetup-bin
  dpkg-repack gir1.2-timezonemap-1.0 gir1.2-xkl-1.0 grub-common
  kde-window-manager kinit kio kpackagetool5 kwayland-data kwin-common
  kwin-data kwin-x11 libdebian-installer4 libkdecorations2-5v5
  libkdecorations2private5v5 libkf5activities5 libkf5attica5
  libkf5completion-data libkf5completion5 libkf5declarative-data
  libkf5declarative5 libkf5doctools5 libkf5globalaccel-data libkf5globalaccel5
  libkf5globalaccelprivate5 libkf5idletime5 libkf5jobwidgets-data
  libkf5jobwidgets5 libkf5kcmutils-data libkf5kcmutils5 libkf5kiocore5
  libkf5kiontlm5 libkf5kiowidgets5 libkf5newstuff-data libkf5newstuff5
  libkf5newstuffcore5 libkf5package-data libkf5package5 libkf5plasma5
  libkf5quickaddons5 libkf5solid5 libkf5solid5-data libkf5sonnet5-data
  libkf5sonnetcore5 libkf5sonnetui5 libkf5textwidgets-data libkf5textwidgets5
  libkf5waylandclient5 libkf5waylandserver5 libkf5xmlgui-bin libkf5xmlgui-data
  libkf5xmlgui5 libkscreenlocker5 libkwin4-effect-builtins1 libkwineffects11
  libkwinglutils11 libkwinxrenderutils11 libqgsttools-p1 libqt5designer5
  libqt5help5 libqt5multimedia5 libqt5multimedia5-plugins
  libqt5multimediaquick-p5 libqt5multimediawidgets5 libqt5positioning5
  libqt5qml5 libqt5quick5 libqt5quickwidgets5 libqt5sensors5 libqt5webchannel5
  libqt5webkit5 libxcb-composite0 libxcb-cursor0 libxcb-damage0 os-prober
  python3-dbus.mainloop.pyqt5 python3-icu python3-pam python3-pyqt5
  python3-pyqt5.qtsvg python3-pyqt5.qtwebkit
  qml-module-org-kde-kquickcontrolsaddons qml-module-qtmultimedia
  qml-module-qtquick2 rdate tasksel tasksel-data
Use 'sudo apt autoremove' to remove them.
0 to upgrade, 0 to newly install, 0 to remove and 0 not to upgrade.
marty@marty-desktop:~$ curl -s https://raw.githubusercontent.com/ros/rosdistro/master/ros.asc | sudo apt-key add -
OK
marty@marty-desktop:~$ 
'''

fishworx, 2019-12-09:
Add the ROS 2 apt repository

'''
marty@marty-desktop:~$ sudo sh -c 'echo "deb [arch=amd64,arm64] http://packages.ros.org/ros2/ubuntu `lsb_release -cs` main" > /etc/apt/sources.list.d/ros2-latest.list'
marty@marty-desktop:~$ 
'''

fishworx, 2019-12-09:
Install development tools and ROS tools

'''
marty@marty-desktop:~$ sudo apt update && sudo apt install -y \
>   build-essential \
>   cmake \
>   git \
>   python3-colcon-common-extensions \
>   python3-pip \
>   python-rosdep \
>   python3-vcstool \
>   wget
Get:1 file:/var/cuda-repo-10-0-local-10.0.326  InRelease
Ign:1 file:/var/cuda-repo-10-0-local-10.0.326  InRelease
Get:2 file:/var/visionworks-repo  InRelease
Ign:2 file:/var/visionworks-repo  InRelease          
Get:3 file:/var/visionworks-sfm-repo  InRelease      
Ign:3 file:/var/visionworks-sfm-repo  InRelease                                
Get:4 file:/var/visionworks-tracking-repo  InRelease                           
Ign:4 file:/var/visionworks-tracking-repo  InRelease                           
Get:5 file:/var/cuda-repo-10-0-local-10.0.326  Release [574 B]                 
Get:6 file:/var/visionworks-repo  Release [1,999 B]                            
Get:7 file:/var/visionworks-sfm-repo  Release [2,003 B]                        
Get:5 file:/var/cuda-repo-10-0-local-10.0.326  Release [574 B]                 
Get:8 file:/var/visionworks-tracking-repo  Release [2,008 B]                   
Hit:9 http://international.download.nvidia.com/jetson/repos/common r32 InRelease
Get:6 file:/var/visionworks-repo  Release [1,999 B]                            
Hit:10 http://international.download.nvidia.com/jetson/repos/t210 r32 InRelease
Get:7 file:/var/visionworks-sfm-repo  Release [2,003 B]                        
Get:8 file:/var/visionworks-tracking-repo  Release [2,008 B]                   
Hit:12 http://ports.ubuntu.com/ubuntu-ports bionic InRelease                   
Hit:13 http://packages.ros.org/ros2/ubuntu bionic InRelease               
Hit:15 http://ports.ubuntu.com/ubuntu-ports bionic-updates InRelease
Hit:18 http://ports.ubuntu.com/ubuntu-ports bionic-backports InRelease
Hit:19 http://ports.ubuntu.com/ubuntu-ports bionic-security InRelease
Reading package lists... Done                       
Building dependency tree       
Reading state information... Done
All packages are up to date.
Reading package lists... Done
Building dependency tree       
Reading state information... Done
build-essential is already the newest version (12.4ubuntu1).
cmake is already the newest version (3.10.2-1ubuntu2.18.04.1).
git is already the newest version (1:2.17.1-1ubuntu0.4).
wget is already the newest version (1.19.4-1ubuntu2.2).
python3-pip is already the newest version (9.0.1-2.3~ubuntu1.18.04.1).
python-rosdep is already the newest version (0.17.1-1).
python3-colcon-common-extensions is already the newest version (0.2.1-1).
python3-vcstool is already the newest version (0.2.4-1).
The following packages were automatically installed and are no longer required:
  apt-clone archdetect-deb bogl-bterm busybox-static cryptsetup-bin
  dpkg-repack gir1.2-timezonemap-1.0 gir1.2-xkl-1.0 grub-common
  kde-window-manager kinit kio kpackagetool5 kwayland-data kwin-common
  kwin-data kwin-x11 libdebian-installer4 libkdecorations2-5v5
  libkdecorations2private5v5 libkf5activities5 libkf5attica5
  libkf5completion-data libkf5completion5 libkf5declarative-data
  libkf5declarative5 libkf5doctools5 libkf5globalaccel-data libkf5globalaccel5
  libkf5globalaccelprivate5 libkf5idletime5 libkf5jobwidgets-data
  libkf5jobwidgets5 libkf5kcmutils-data libkf5kcmutils5 libkf5kiocore5
  libkf5kiontlm5 libkf5kiowidgets5 libkf5newstuff-data libkf5newstuff5
  libkf5newstuffcore5 libkf5package-data libkf5package5 libkf5plasma5
  libkf5quickaddons5 libkf5solid5 libkf5solid5-data libkf5sonnet5-data
  libkf5sonnetcore5 libkf5sonnetui5 libkf5textwidgets-data libkf5textwidgets5
  libkf5waylandclient5 libkf5waylandserver5 libkf5xmlgui-bin libkf5xmlgui-data
  libkf5xmlgui5 libkscreenlocker5 libkwin4-effect-builtins1 libkwineffects11
  libkwinglutils11 libkwinxrenderutils11 libqgsttools-p1 libqt5designer5
  libqt5help5 libqt5multimedia5 libqt5multimedia5-plugins
  libqt5multimediaquick-p5 libqt5multimediawidgets5 libqt5positioning5
  libqt5qml5 libqt5quick5 libqt5quickwidgets5 libqt5sensors5 libqt5webchannel5
[... 138 lines cut ...]
  libqt5qml5 libqt5quick5 libqt5quickwidgets5 libqt5sensors5 libqt5webchannel5
  libqt5webkit5 libxcb-composite0 libxcb-cursor0 libxcb-damage0 os-prober
  python3-dbus.mainloop.pyqt5 python3-icu python3-pam python3-pyqt5
  python3-pyqt5.qtsvg python3-pyqt5.qtwebkit
  qml-module-org-kde-kquickcontrolsaddons qml-module-qtmultimedia
  qml-module-qtquick2 rdate tasksel tasksel-data
Use 'sudo apt autoremove' to remove them.
Recommended packages:
  libboost-regex-dev libboost-dev libboost-date-time-dev
The following NEW packages will be installed:
  libasio-dev
0 to upgrade, 1 to newly install, 0 to remove and 0 not to upgrade.
Need to get 240 kB of archives.
After this operation, 2,866 kB of additional disk space will be used.
Get:1 http://ports.ubuntu.com/ubuntu-ports bionic/universe arm64 libasio-dev all 1:1.10.8-1 [240 kB]
Fetched 240 kB in 2s (109 kB/s)       
debconf: delaying package configuration, since apt-utils is not installed
Selecting previously unselected package libasio-dev.
(Reading database ... 153136 files and directories currently installed.)
Preparing to unpack .../libasio-dev_1%3a1.10.8-1_all.deb ...
Unpacking libasio-dev (1:1.10.8-1) ...
Setting up libasio-dev (1:1.10.8-1) ...
marty@marty-desktop:~$ 

'''

fishworx, 2019-12-09:
Get ROS 2 code

'''
marty@marty-desktop:~$ mkdir -p ~/ros2_eloquent/src
marty@marty-desktop:~$ cd ~/ros2_eloquent
marty@marty-desktop:~/ros2_eloquent$ wget https://raw.githubusercontent.com/ros2/ros2/eloquent/ros2.repos
--2019-12-09 12:51:08--  https://raw.githubusercontent.com/ros2/ros2/eloquent/ros2.repos
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 151.101.96.133
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|151.101.96.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 10962 (11K) [text/plain]
Saving to: 'ros2.repos.1’

ros2.repos.1        100%[===================>]  10.71K  --.-KB/s    in 0.003s  

2019-12-09 12:51:08 (3.66 MB/s) - 'ros2.repos.1’ saved [10962/10962]

marty@marty-desktop:~/ros2_eloquent$ vcs import src < ros2.repos
...............................................................................................
=== src/ament/ament_cmake (git) ===

Already on 'eloquent'
Your branch is up to date with 'origin/eloquent'.
=== src/ament/ament_index (git) ===

Already on 'eloquent'
Your branch is up to date with 'origin/eloquent'.
=== src/ament/ament_lint (git) ===

Already on 'eloquent'
Your branch is up to date with 'origin/eloquent'.
=== src/ament/ament_package (git) ===

Already on 'eloquent'
Your branch is up to date with 'origin/eloquent'.
=== src/ament/googletest (git) ===

Already on 'ros2'
Your branch is up to date with 'origin/ros2'.
=== src/ament/uncrustify_vendor (git) ===

Already on 'eloquent'
Your branch is up to date with 'origin/eloquent'.
=== src/eProsima/Fast-CDR (git) ===

HEAD is now at cc27c24 Refs #6203 Remove FastRTPSGen code. (#61)
=== src/eProsima/Fast-RTPS (git) ===
From https://github.com/eProsima/Fast-RTPS
 * [new branch]          feature/participant-whitelist/irobot -> origin/feature/participant-whitelist/irobot
   2dbb53b7f..34337d5c5  irobot_test           -> origin/irobot_test
 * [new branch]          irobot_test_common_event_thread -> origin/irobot_test_common_event_thread
 * [new branch]          irobot_test_whitelist -> origin/irobot_test_whitelist
Already on '1.9.x'
Your branch is up to date with 'origin/1.9.x'.
=== src/eProsima/foonathan_memory_vendor (git) ===

Already on 'master'
Your branch is up to date with 'origin/master'.
=== src/eclipse-cyclonedds/cyclonedds (git) ===

HEAD is now at 08d9c29 Add DOMAIN_ID, DOMAIN_TAG to config, discovery
=== src/micro-ROS/ros_tracing/ros2_tracing (git) ===
From https://gitlab.com/micro-ROS/ros_tracing/ros2_tracing
   b10e1e6..f84659f  master     -> origin/master
HEAD is now at 866dada 0.2.10
=== src/osrf/osrf_pycommon (git) ===

Already on 'eloquent'
Your branch is up to date with 'origin/eloquent'.
=== src/osrf/osrf_testing_tools_cpp (git) ===
[... 315 lines cut ...]
Already on 'eloquent'
Your branch is up to date with 'origin/eloquent'.
=== src/ros2/tlsf (git) ===

Already on 'eloquent'
Your branch is up to date with 'origin/eloquent'.
=== src/ros2/unique_identifier_msgs (git) ===

Already on 'eloquent'
Your branch is up to date with 'origin/eloquent'.
=== src/ros2/urdf (git) ===

Already on 'eloquent'
Your branch is up to date with 'origin/eloquent'.
=== src/ros2/urdfdom (git) ===

Already on 'eloquent'
Your branch is up to date with 'origin/eloquent'.
=== src/ros2/yaml_cpp_vendor (git) ===

Already on 'eloquent'
Your branch is up to date with 'origin/eloquent'.
marty@marty-desktop:~/ros2_eloquent$ 

'''

fishworx, 2019-12-09:
ERROR HERE.... 'src' does not exist....

Install dependencies using rosdep

'''

marty@marty-desktop:~$ sudo rosdep init
[sudo] password for marty: 
ERROR: default sources list file already exists:
	/etc/ros/rosdep/sources.list.d/20-default.list
Please delete if you wish to re-initialize
marty@marty-desktop:~$ rosdep update
reading in sources list data from /etc/ros/rosdep/sources.list.d
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/osx-homebrew.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/base.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/python.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/ruby.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/releases/fuerte.yaml
Query rosdistro index https://raw.githubusercontent.com/ros/rosdistro/master/index-v4.yaml
Skip end-of-life distro "ardent"
Skip end-of-life distro "bouncy"
Add distro "crystal"
Add distro "dashing"
Add distro "eloquent"
Add distro "foxy"
Skip end-of-life distro "groovy"
Skip end-of-life distro "hydro"
Skip end-of-life distro "indigo"
Skip end-of-life distro "jade"
Add distro "kinetic"
Skip end-of-life distro "lunar"
Add distro "melodic"
Add distro "noetic"
updated cache in /home/marty/.ros/rosdep/sources.cache
marty@marty-desktop:~$ rosdep install --from-paths src --ignore-src --rosdistro eloquent -y --skip-keys "console_bridge fastcdr fastrtps libopensplice67 libopensplice69 rti-connext-dds-5.3.1 urdfdom_headers"
given path 'src' does not exist
marty@marty-desktop:~$ 
'''

fishworx, 2019-12-09:
ERROR fixed... went to correct directory then reran

'''
marty@marty-desktop:~$ cd ~/ros2_eloquent
marty@marty-desktop:~/ros2_eloquent$ rosdep install --from-paths src --ignore-src --rosdistro eloquent -y --skip-keys "console_bridge fastcdr fastrtps libopensplice67 libopensplice69 rti-connext-dds-5.3.1 urdfdom_headers"
#All required rosdeps installed successfully

'''

fishworx, 2019-12-09:
RTI Connext (version 5.3.1, amd64 only)- NEW ERROR...

'''
marty@marty-desktop:~/ros2_eloquent$ sudo apt install -q -y \
>     rti-connext-dds-5.3.1  # from packages.ros.org/ros2/ubuntu
Reading package lists...
Building dependency tree...
Reading state information...
rti-connext-dds-5.3.1 is already the newest version (0.0.0-0).
The following packages were automatically installed and are no longer required:
  apt-clone archdetect-deb bogl-bterm busybox-static cryptsetup-bin
  dpkg-repack gir1.2-timezonemap-1.0 gir1.2-xkl-1.0 grub-common
  kde-window-manager kinit kio kpackagetool5 kwayland-data kwin-common
  kwin-data kwin-x11 libdebian-installer4 libkdecorations2-5v5
  libkdecorations2private5v5 libkf5activities5 libkf5attica5
  libkf5completion-data libkf5completion5 libkf5declarative-data
  libkf5declarative5 libkf5doctools5 libkf5globalaccel-data libkf5globalaccel5
  libkf5globalaccelprivate5 libkf5idletime5 libkf5jobwidgets-data
  libkf5jobwidgets5 libkf5kcmutils-data libkf5kcmutils5 libkf5kiocore5
  libkf5kiontlm5 libkf5kiowidgets5 libkf5newstuff-data libkf5newstuff5
  libkf5newstuffcore5 libkf5package-data libkf5package5 libkf5plasma5
  libkf5quickaddons5 libkf5solid5 libkf5solid5-data libkf5sonnet5-data
  libkf5sonnetcore5 libkf5sonnetui5 libkf5textwidgets-data libkf5textwidgets5
  libkf5waylandclient5 libkf5waylandserver5 libkf5xmlgui-bin libkf5xmlgui-data
  libkf5xmlgui5 libkscreenlocker5 libkwin4-effect-builtins1 libkwineffects11
  libkwinglutils11 libkwinxrenderutils11 libqgsttools-p1 libqt5designer5
  libqt5help5 libqt5multimedia5 libqt5multimedia5-plugins
  libqt5multimediaquick-p5 libqt5multimediawidgets5 libqt5positioning5
  libqt5qml5 libqt5quick5 libqt5quickwidgets5 libqt5sensors5 libqt5webchannel5
  libqt5webkit5 libxcb-composite0 libxcb-cursor0 libxcb-damage0 os-prober
  python3-dbus.mainloop.pyqt5 python3-icu python3-pam python3-pyqt5
  python3-pyqt5.qtsvg python3-pyqt5.qtwebkit
  qml-module-org-kde-kquickcontrolsaddons qml-module-qtmultimedia
  qml-module-qtquick2 rdate tasksel tasksel-data
Use 'sudo apt autoremove' to remove them.
0 to upgrade, 0 to newly install, 0 to remove and 0 not to upgrade.
marty@marty-desktop:~/ros2_eloquent$ cd /opt/rti.com/rti_connext_dds-5.3.1/resource/scripts && source ./rtisetenv_x64Linux3gcc5.4.0.bash; cd -
bash: cd: /opt/rti.com/rti_connext_dds-5.3.1/resource/scripts: No such file or directory
/home/marty
marty@marty-desktop:~$ cd
marty@marty-desktop:~$ cd /opt/rti.com/rti_connext_dds-5.3.1/resource/scripts && source ./rtisetenv_x64Linux3gcc5.4.0.bash; cd -
bash: cd: /opt/rti.com/rti_connext_dds-5.3.1/resource/scripts: No such file or directory
/home/marty
marty@marty-desktop:~$ 
'''

fishworx, 2019-12-09:
Build the code in the workspace (12hr 22mm build time)

'''
marty@marty-desktop:~$ cd ~/ros2_eloquent/
marty@marty-desktop:~/ros2_eloquent$ colcon build --symlink-install
Starting >>> ament_package
Starting >>> ament_lint
Starting >>> fastcdr                                                           
Starting >>> foonathan_memory_vendor
Finished <<< fastcdr [6.12s]                                   
Starting >>> poco_vendor                                       
Finished <<< foonathan_memory_vendor [6.32s]                   
Starting >>> fastrtps
Finished <<< ament_package [6.61s]                             
Starting >>> ament_cmake_core
Finished <<< poco_vendor [1.06s]                            
Starting >>> cyclonedds                                     
Finished <<< ament_cmake_core [1.08s]                       
Starting >>> ament_cmake_libraries
Finished <<< ament_lint [8.09s]                             
Starting >>> ament_cmake_export_libraries
Finished <<< ament_cmake_libraries [1.01s]                      
Starting >>> ament_cmake_include_directories
Finished <<< ament_cmake_export_libraries [0.92s]               
Starting >>> ament_cmake_python                                 
Finished <<< ament_cmake_include_directories [0.96s]            
Starting >>> ament_cmake_export_dependencies                    
Finished <<< cyclonedds [2.76s]                                 
Starting >>> ament_cmake_export_definitions                      
Finished <<< ament_cmake_python [1.05s]
Starting >>> ament_cmake_test                                    
Finished <<< ament_cmake_export_definitions [0.97s]              
Starting >>> ament_cmake_export_include_directories
Finished <<< ament_cmake_export_dependencies [1.31s]             
Starting >>> ament_cmake_export_link_flags
Finished <<< ament_cmake_test [1.23s]                            
Starting >>> ament_cmake_version                                    
Finished <<< ament_cmake_export_include_directories [1.51s]         
Starting >>> ament_cmake_export_interfaces                          
Finished <<< ament_cmake_export_link_flags [1.64s]                  
Starting >>> ament_cmake_target_dependencies                        
Finished <<< ament_cmake_version [1.76s]                            
Starting >>> ament_flake8                                           
Finished <<< ament_cmake_export_interfaces [6.97s]                   
Starting >>> ament_cppcheck                                          
Finished <<< ament_cmake_target_dependencies [9.65s]                 
Starting >>> ament_cmake                                             
Finished <<< ament_flake8 [12.2s]                                     
Starting >>> ament_pep257                                             
Finished <<< ament_cppcheck [11.3s]                                   
Starting >>> tinydir_vendor                                           
Finished <<< ament_cmake [9.35s]                                      
Starting >>> uncrustify_vendor                                        
Finished <<< ament_pep257 [10.4s]                                     
Starting >>> ament_copyright
Finished <<< uncrustify_vendor [8.05s]                                
Starting >>> ament_index_python                                       
Finished <<< ament_copyright [13.5s]                                      
Starting >>> ament_lint_cmake                                             
Finished <<< ament_index_python [15.4s]                                    
Starting >>> ament_xmllint                                                 
Finished <<< ament_lint_cmake [10.9s]                                      
Finished <<< tinydir_vendor [29.6s]                                        
Starting >>> ament_cmake_lint_cmake                                        
Starting >>> ament_cpplint
Finished <<< ament_xmllint [8.82s]                                         
Starting >>> ament_uncrustify                                              
Finished <<< ament_cpplint [7.23s]                                             
Finished <<< ament_cmake_lint_cmake [7.36s]                                    
Starting >>> ament_cmake_copyright                                             
[... 1195 lines cut ...]
[Processing: rviz_default_plugins]      
[Processing: rviz_default_plugins]      
[Processing: rviz_default_plugins]] ... 
[Processing: rviz_default_plugins]      
[Processing: rviz_default_plugins]      
[Processing: rviz_default_plugins]      
[Processing: rviz_default_plugins]      
[Processing: rviz_default_plugins]      
[Processing: rviz_default_plugins]      
[Processing: rviz_default_plugins]      
[Processing: rviz_default_plugins]      
[Processing: rviz_default_plugins]      
[Processing: rviz_default_plugins]      
[Processing: rviz_default_plugins]      
[Processing: rviz_default_plugins]      
Finished <<< rviz_default_plugins [9h 54min 53s]
Starting >>> rviz2
[Processing: rviz2]                                           
Finished <<< rviz2 [47.9s]                                       

Summary: 298 packages finished [12h 22min 56s]
  6 packages had stderr output: qt_gui_cpp rmw_connext_cpp rmw_connext_shared_cpp ros1_bridge rosidl_typesupport_connext_c rosidl_typesupport_connext_cpp
marty@marty-desktop:~/ros2_eloquent$ 

'''

fishworx, 2019-12-09:
resot fof compile commands okay..... can now run examples... only error was the directory changed to root at 'Install dependencies using rosdep'.... just add the command 'cd ~/ros2_eloquent' before 'sudo rosdep init'

closed 2019-12-09 by fishworx