ros2/ros2_documentation #487: Choco does not install python debug binaries

nbbrooks, 2019-12-31:
When following the ["Installing ROS 2 on Windows"](https://index.ros.org/doc/ros2/Installation/Eloquent/Windows-Install-Binary/) instructions for Eloquent for ROS2 debug libraries, trying to run ros commands results in an error that "python_d" cannot be found. 

This is not surprising as choco does not install the python debug binaries. I wasn't familiar with choco before trying this, but after a brief search, I couldn't find another choco package that would achieve this.

If someone knows of a way to have choco install this I can make a PR to the documentation for that, otherwise I will make a PR that tells people to download the python 3.7.5 installer to modify their installation to include the debug binaries.

wjwwood, 2020-01-09:
That is mentioned here: https://index.ros.org/doc/ros2/Installation/Eloquent/Windows-Development-Setup/#extra-stuff-for-debug-mode as we assume you're not running debug unless you're also building from source.

mjcarroll, 2020-01-09:
We have found that the Python Debug story on Windows is a bit difficult.  As far as I know, choco provides no mechanism for installing the debug python interpreter, which is why we have the additional installation instructions as @wjwwood mentions above.

brawner, 2020-01-14:
python_d and the required debug symbols can also be installed through Python's regular installer. I have had trouble with 3.8, but 3.7.6 seems to work. https://www.python.org/downloads/windows/

When installing, you have to select customized install, and when clicking next it will give you the option to install the the debug binary and symbols.

nbbrooks, 2020-01-14:
OK, I'll make a PR that copies over the documentation in the source install instructions William linked to the debug library install instructions.

nbbrooks, 2020-01-14:
Actually, I think the PR for the debug library install process will instruct people to use the Python 3.7.3 installer (installing the debug symbols) and remove the choco python installation bits. The source install "Extra stuff for Debug mode" documentation builds python from source, which isn't in line with the objective of the library releases.