ros2/ros2_documentation #488: ROS2 release libraries for Windows: Failed to load shared library of rmw implementation

nbbrooks, 2019-12-31:
When following the ["Installing ROS 2 on Windows"](https://index.ros.org/doc/ros2/Installation/Eloquent/Windows-Install-Binary/) instructions for Eloquent for ROS2 release libraries, trying to run ros commands (ex: ros2 run demo_nodes_py listener) results in the following error:

>RuntimeError: Failed to initialize init_options: failed to load shared library of rmw implementation. Exception: Cannot load library: C:\dev\ros2_eloquent\bin/rmw_fastrtps_cpp.dll, at C:\J\workspace\packaging_windows\ws\src\ros2\rmw_implementation\rmw_implementation\src\functions.cpp:137, at C:\J\workspace\packaging_windows\ws\src\ros2\rcl\rcl\src\rcl\init_options.c:55 

Note that I used the install directory in the tutorial - the C:\J path must be a backup location set from when the binary was built.

I was unable to fix this but tried a couple things:
- "Unblocked" all the DLLs in the ros2_eloquent directory (using this command https://www.404techsupport.com/2016/06/24/unblock-files-powershell)
- Ran [https://github.com/lucasg/Dependencies](Dependencies) to try and find any missing DLL dependencies for rmw_fastrtps_cpp.dll, but I did not see anything

I have seen this issue in many other places with no successful resolution (other than installing ROS2 from source) and wanted to put the issue here and confirm it is still present in Eloquent.

seanyen, 2020-01-07:
@nbbrooks Could you share what windows build you are on? Perhaps a screenshot from `winver` command would help. Also what's the ROS2 build you are using?

nbbrooks, 2020-01-11:
My Windows build is
>Microsoft Windows Version 1903 (OS Build 18362.535)

Path is
> C:\Users\nathanbrooks>echo %PATH%                                                                                       C:\Python37\Scripts\;C:\Python37\;C:\ProgramData\DockerDesktop\version-bin;C:\Program Files\Docker\Docker\Resources\bin;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\Program Files (x86)\Windows Kits\8.1\Windows Performance Toolkit\;C:\ProgramData\chocolatey\bin;C:\ProgramData\chocolatey\lib\cunit\lib;C:\ProgramData\chocolatey\lib\tinyxml2\lib;C:\ProgramData\chocolatey\lib\log4cxx\lib;C:\Program Files\eProsima\fastrtps 1.9.3\bin;C:\Program Files\eProsima\fastrtps 1.9.3\bin\x64Win64VS2017;C:\Program Files\Git\cmd;C:\Program Files\Cppcheck;;C:\Program Files\Microsoft VS Code\bin;C:\Program Files (x86)\Windows Kits\10\Windows Performance Toolkit\;C:\Users\nathanbrooks\AppData\Local\Microsoft\WindowsApps;C:\Program Files\OpenSSL-Win64\bin;C:\Program Files\CMake\bin;C:\Program Files\Git\cmd;C:\xmllint\bin;         

ROS2 build
> Eloquent Elusor 20191122 https://github.com/ros2/ros2/releases/tag/release-eloquent-20191122

martimorta, 2020-01-17:
Same here, I haven't done the DLL unblocking neither check for dependencies though.
Windows build:

> 1909 OS Build 18363.535

Path

> C:\dev\ros2_eloquent\opt\yaml_cpp_vendor\bin;C:\dev\ros2_eloquent\opt\rviz_ogre_vendor\bin;C:\dev\ros2_eloquent\opt\rviz_assimp_vendor\bin;C:\dev\ros2_eloquent\opt\libcurl_vendor\bin;C:\dev\ros2_eloquent\Scripts;C:\dev\ros2_eloquent\bin;C:\Python37\Scripts\;C:\Python37\;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\Program Files\Microsoft VS Code\bin;C:\Program Files\Git\cmd;C:\Program Files\nodejs\;C:\Program Files\Git LFS;C:\ProgramData\chocolatey\bin;C:\Users\marti\AppData\Local\Microsoft\WindowsApps;C:\Users\marti\AppData\Local\Programs\Microsoft VS Code Insiders\bin;C:\Users\marti\AppData\Roaming\npm;C:\Program Files\OpenSSL-Win64\bin;C:\opencv-3.4.6\x64\vc16\bin;C:\Program Files\CMake\bin;

ROS2 build

> Eloquent Elusor 20191122 https://github.com/ros2/ros2/releases/tag/release-eloquent-20191122

nbbrooks, 2020-01-30:
I will further test and issue a PR, but I found this issue over in ros2
https://github.com/ros2/ros2/issues/844

On this machine I installed Eloquent Patch Release 1 and Win64 OpenSSL v1.0.2u Light and am no longer getting the rmw implementation error and the talker and demo listener nodes successfully communicate.

martimorta, 2020-02-01:
Thanks for that @nbbrooks, sadly I still find the problem, these are steps I have tried
1. Remove /dev/ros2_eloquent and "install" ros2-eloquent-20200124-windows-release-amd64
2. Remove OpenSSL 1.1.0
3. Install OpenSSL 1.0.2u Light 64bit and try again
4. Install OpenSSL 1.0.2u 64bit and try again
5. Set RMW_IMPLEMENTATION=rmw_fastrtps_cpp and try again

The errors and system info are shown below.
Thanks!

For 3 and 4 I have the following errors in the local_setup
```
C:\Users\marti>call C:\dev\ros2_eloquent\local_setup.bat
[connext_cmake_module] Warning: The location at which Connext was found when the workspace was built [[C:\Program Files\rti_connext_dds-5.3.1]] does not point to a valid directory, and the NDDSHOME environment variable has not been set. Support for Connext will not be available.
[opensplice_cmake_module] Warning: The location at which OpenSplice was found when the workspace was built [[C:\dev\opensplice\HDE\x86_64.win64\]] does not point to a valid directory, and the OSPL_HOME environment variable has not been set. Support for OpenSplice will not be available.
```
And the one about rmw
``` 
C:\Users\marti>ros2 run demo_nodes_py listener
Traceback (most recent call last):
  File "C:\dev\ros2_eloquent\lib\demo_nodes_py\listener-script.py", line 11, in <module>
    load_entry_point('demo-nodes-py==0.8.4', 'console_scripts', 'listener')()
  File "C:\dev\ros2_eloquent\Lib\site-packages\demo_nodes_py\topics\listener.py", line 32, in main
    rclpy.init(args=args)
  File "C:\dev\ros2_eloquent\Lib\site-packages\rclpy\__init__.py", line 73, in init
    return rclpy_implementation.rclpy_init(args if args is not None else sys.argv, context.handle)
RuntimeError: Failed to initialize init_options: failed to load shared library of rmw implementation. Exception: Cannot load library: C:\dev\ros2_eloquent\bin/rmw_fastrtps_cpp.dll, at C:\J\workspace\ci_packaging_windows\ws\src\ros2\rmw_implementation\rmw_implementation\src\functions.cpp:137, at C:\J\workspace\ci_packaging_windows\ws\src\ros2\rcl\rcl\src\rcl\init_options.c:55
```
For 5, (after adding rmw_implementation) the error becomes
```
[ERROR] [rcl]: Error getting RMW implementation identifier / RMW implementation not installed (expected identifier of 'rmw_fastrtps_cpp'), with error message 'failed to load shared library of rmw implementation. Exception: Cannot load library: C:\dev\ros2_eloquent\bin/rmw_fastrtps_cpp.dll, at C:\J\workspace\ci_packaging_windows\ws\src\ros2\rmw_implementation\rmw_implementation\src\functions.cpp:137', exiting with 1.
```

Windows build
>1909 OS Build 18363.535

Path
```
C:\dev\ros2_eloquent\opt\yaml_cpp_vendor\bin;C:\dev\ros2_eloquent\opt\rviz_ogre_vendor\bin;C:\dev\ros2_eloquent\opt\rviz_assimp_vendor\bin;C:\dev\ros2_eloquent\opt\libcurl_vendor\bin;C:\dev\ros2_eloquent\Scripts;C:\dev\ros2_eloquent\bin;C:\Python37\Scripts\;C:\Python37\;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\Program Files\Microsoft VS Code\bin;C:\Program Files\nodejs\;C:\Program Files\Git LFS;C:\ProgramData\chocolatey\bin;C:\Program Files\Cppcheck;C:\Program Files\Git\cmd;;C:\Users\marti\AppData\Local\Microsoft\WindowsApps;C:\Users\marti\AppData\Local\Programs\Microsoft VS Code Insiders\bin;C:\Users\marti\AppData\Roaming\npm;C:\OpenSSL-Win64\bin;C:\opencv-3.4.6\x64\vc16\bin;C:\Program Files\CMake\bin;C:\Program Files\xmllint\bin; 
```

nbbrooks, 2020-02-03:
Hmmm, that is unfortunate. I still don't know exactly what is going on except that the 1.0.2u light x64 version must be installed a dependency the non-light version doesn't, and it does not get removed when uninstalling.
- With a fresh windows install following the current installation instructions installing OpenSSL x64 1.0.2u (non-light) I get the ' failed to load shared library of rmw implementation' error
- If I then uninstall that OpenSSL version and install x64 1.0.2u light I no longer get that error and the talker/listener demo works
- If I then uninstall the light version, the talker/listener demo still works.

Some other notes:
> Set RMW_IMPLEMENTATION=rmw_fastrtps_cpp and try again

I have not done this step

>C:\Users\marti>call C:\dev\ros2_eloquent\local_setup.bat
[connext_cmake_module] Warning: The location at which Connext was found when the workspace was built [[C:\Program Files\rti_connext_dds-5.3.1]] does not point to a valid directory, and the NDDSHOME environment variable has not been set. Support for Connext will not be available.
[opensplice_cmake_module] Warning: The location at which OpenSplice was found when the workspace was built [[C:\dev\opensplice\HDE\x86_64.win64\]] does not point to a valid directory, and the OSPL_HOME environment variable has not been set. Support for OpenSplice will not be available.

I also get this, but believe it is "normal" as I did not install Connext or OpenSlice. I'm not sure why it expects to find them by default, but that should get addressed.

mjcarroll, 2020-02-04:
> C:\Users\marti>call C:\dev\ros2_eloquent\local_setup.bat
[connext_cmake_module] Warning: The location at which Connext was found when the workspace was built [[C:\Program Files\rti_connext_dds-5.3.1]] does not point to a valid directory, and the NDDSHOME environment variable has not been set. Support for Connext will not be available.
[opensplice_cmake_module] Warning: The location at which OpenSplice was found when the workspace was built [[C:\dev\opensplice\HDE\x86_64.win64]] does not point to a valid directory, and the OSPL_HOME environment variable has not been set. Support for OpenSplice will not be available.

That is expected if you don't have OpenSplice or Connext installed.

mjcarroll, 2020-02-04:
The way that I normally go about debugging these issues is with https://github.com/lucasg/Dependencies

It will let you see if you are depending on a DLL that isn't currently in the path (either via path mis-configuration or it not being installed)

martimorta, 2020-02-04:
Hi, thanks for testing this. I just tried removing OpenSSL 1.0.2u and installing the light version but it didn't help in my case. I also tried "dependencies.exe" with the listener.exe but I am probably doing it wrong since it only shows some windows' kernel dll's, though some of them seem to be missing.

[image]

mjcarroll, 2020-02-04:
My hunch is that those come from a VC Redistributable?

Did you do the choco install: `choco install -y vcredist2013 vcredist140`

Or perhaps @seanyen may have some insight.

nbbrooks, 2020-02-06:
I also looked at using Dependencies but didn't get any useful information out of it (could have been user error though).

I did have a conversation with @seanyen and @ooeygui about this as I was putting together this issue. Lou suggested trying loader snaps to find the dll loading failure, but I have not gotten around to testing that:

> When debugging a dll load failure, Windows provides a diagnostics feature called Loader Snaps. You can enable this on a per executable name.
> 
> There is a UI tool for this in the Debugging Tools:
> https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/
> (There is a faster way to do this through the Windows Registry; it’s a bit more complex to get right and easy to get badly wrong)
> 
> Once you install this, run gflags.exe (You can press the Windows key and R to open a run dialog).
> 
> Type the exe name for your host. (like ros2.exe), then press the tab key. This will enable the checkboxes. Then select Show Loader Snaps, then press Ok to close the dialog.
> [image]
> 
> When you debug the executable, it will spew tons of data. One of the line items will be the failing export.

Side note: I was helping someone set up a ROS2 project on Windows the other day, and ran into issues with the Eigen cmake config file being found when building the workspace. In the end, I believe this was because they forgot to add CMake to the Path - we added it later, but after installation of all the chocolatey packages. The error was probably because either
- not having cmake in PATH when installing chocolatey packages prevented cmake from checking chocolatey's folders for cmake configs
- chocolatey had an error of some sort during installation/registry entry (we had to manually delete registry entries to reinstall the chocolatey packages)
I can't think of a reason why this would be the case for this dll failure, but in light of @mjcarroll comment it might be worth reinstalling all the chocolatey pakcages (--force)

martimorta, 2020-02-10:
I found out I had cmake tools for VS2017, so I removed them, VS2019 didn't have them though. I tried reinstalling all choco packages but the error still appears. I think I will have to remove everything and reinstall again. If I find any solution I will post it here. Thanks anyway!

nbbrooks, 2020-02-10:
I don't think having VS cmake tools installed breaks things, but it can complicate figuring out why things don't work (such as if chocolatey's cmake isn't in path). When you re-install VS2019, make sure to de-select "C++ Cmake tools for Windows" when checking "Desktop development with C++". I have a PR to update the documentation pending.

[image]

nuclearsandwich, 2020-02-27:
With #514 merged (Thanks @nbbrooks!) I think this issue might be wrapped up.

We had recommended not using the CMake version installed with Visual Studio 2019 as the initial distribution of CMake with Visual Studio didn't actually include support for Visual Studio 2019. Although I expect that has been resolved.

nuclearsandwich, 2020-03-05:
I'll go ahead and close this if a participant feels like more is required we can open a new issue or reopen this one.

closed 2020-03-05 by nuclearsandwich

MahyarBayran, 2021-04-20:
I had the same problem. I reinstalled a lot of stuff and did some stuff mentioned here. At the end, what fixed my problem was installing from [ROS 2 Build Installation from aka.ms/ros](https://docs.ros.org/en/foxy/Installation/Windows-Install-Binary.html#alternative-ros-2-build-installation-from-aka-ms-ros), instead of [Downloading ROS 2](https://docs.ros.org/en/foxy/Installation/Windows-Install-Binary.html#downloading-ros-2)

bala4901, 2023-01-08:
Hi, 

I found another solution for by [By Installation OpenSSL](https://qiita.com/comocc/items/f9dd3130acdff798d4f9).

1, Download and Install  [Win64 OpenSSL v1.1.1](https://slproweb.com/products/Win32OpenSSL.html). Note: can find the MSI at bottom of the page
2. `setx -m OPENSSL_CONF "C:\Program Files\OpenSSL-Win64\bin\openssl.cfg"`

Then try:
`ros2 run demo_nodes_py listener `

Result:

> [INFO] [1673152950.727193900] [talker]: Publishing: "Hello World: 0"
> [INFO] [1673152951.569101000] [talker]: Publishing: "Hello World: 1"
> [INFO] [1673152952.570804600] [talker]: Publishing: "Hello World: 2"

waytoashutosh, 2025-04-15:
What worked for me was to first uninstall C:\Program Files\OpenSSL-Win64 and then 
reinstall using the windows documentation.

 https://docs.ros.org/en/humble/Installation/Windows-Install-Binary.html#id13

`choco install -y openssl --version 1.1.1.2100`