ros2/ros2_documentation #508: Pyhton Simple Service:

matthijsramlab, 2020-01-29:
In the package.xml package the following line is created when I create the package: 

`<buildtool_depend>ament_python</buildtool_depend>`

When I run `sudo rosdep install -i --from-path src --rosdistro eloquent -y` it gives me the following error:

```
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
py_srvcli: Cannot locate rosdep definition for [ament_python]
```

What did I do wrong? Does the package need to depend on the ament_pyhon, it does build without any problems

dirk-thomas, 2020-01-29:
Assuming you used the `ros2 pkg create` command to generate the Python package that had a bug which was fixed on `master` in ros2/ros2cli#428. The backport to Eloquent is pending ros2/ros2cli#439 and will be part of the next patch release.

You can simply remove the `<buildtool_depend>ament_python</buildtool_depend>` from your `package.xml` file.

closed 2020-01-29 by dirk-thomas

felixdivo, 2021-04-21:
Should it also be removed from here: https://docs.ros.org/en/rolling/Tutorials/Creating-Your-First-ROS2-Package.html#customize-package-xml ?

clalancette, 2021-04-21:
> Should it also be removed from here: https://docs.ros.org/en/rolling/Tutorials/Creating-Your-First-ROS2-Package.html#customize-package-xml ?

Yep, definitely.  Do you mind opening a PR to do that?

felixdivo, 2021-04-21:
Done in #1393.

peci1, 2025-03-09:
Shouldn't there be `<buildtool_depend>python3-setuptools</buildtool_depend>` ? Or who makes sure setuptools is installed?