ros2/ros2_documentation #509: Access Control

tylerjw, 2020-01-30:
This is in reference to the lines here: https://index.ros.org/doc/ros2/Contributing/Developer-Guide/#access-control

```Access Control
Drop requirement for all class members to be private and therefore require accessors
* rationale: this is overly constraining for user API design
* we should prefer private members, only making them public when they are needed
* we should consider using accessors before choosing to allow direct member access
* we should have a good reason for allowing direct member access, other than because it is convenient for us
```

I have seen several cases public and protected member variables were rampant and led to some really hard to read/understand/modify/use/debug code.  For this big of a change to the Google style guide (https://google.github.io/styleguide/cppguide.html#Access_Control) this should come with some examples of why the authors think this is a useful modification of the Google style guide.

For example, what are some user APIs that are better because they have public or protected member variables?  Are there examples of other mature, well respected C++ libraries that have this rule and examples where their API is better for having many public and protected member variables?

I'm not asking for this to change.  I just want examples of why this is "overly constraining" and worth the tradeoffs.

tylerjw, 2020-02-06:
@tfoote do you have an opinion on this?

tylerjw, 2020-02-06:
The primary downside to Google's rule for this is you have to write some more boilerplate code and spend some time reasoning about architecture decisions.

The upside to writing this boilerplate code (setters and accessors) is that you have to address the scope and mutability and invariance of state.  This boilerplate has the upside of making the code easier to reason about and difficult for users of your API to create really annoying concurrency bugs.

"public" state of objects is the default behavior of structs.  If this is the behavior you want using a struct seems like a much nicer way to communicate to your API users how you expect that to be used.  Having some data members be public state of classes seems like a good way to make your API more confusing to users.

"protected" state is sometimes very useful but it has the downside of making it much harder to reason about the invariance of state.  There are also very real performance implications to the virtual table.  A bias towards harder to write but easier to read and implement code against library code seems like a reasonable design principal especially when it comes with performance benefits.   

For these reasons I think this would be reasonable to add to the guide (in place of the "overly constraining" lines):

```
Public class state should be explicitly avoided.
  * Use a struct instead
Protected class state and inheritance should be explicitly avoided.  
  * Rational: diamond inheritance, virtual table lookup performance implications, API simplicity, reasoning about state invariance, etc. 
```

tfoote, 2020-02-07:
You can't switch your class to be a struct without violating the struct guidelines that they are just data containers and don't have methods or private contents.

> "public" state of objects is the default behavior of structs. If this is the behavior you want using a struct seems like a much nicer way to communicate to your API users how you expect that to be [...]

This is one of the reasons why we discourage it. 

Your comments about protected are already in the Google style guide.

An example where this sort of exception might be valuable would be in operating on a large data structure such as an image or map where we want to do in place operations such as convolutions or iterations.

We are dropping the requirement of the restriction but indicating that it should not be chosen unless you have a good reason. This is simply relaxing a hard constraint but still recommending it be followed. Which should not end up with it being "rampant".

closed 2020-03-02 by tfoote