ros2/ros2_documentation #510: Exceptions in destructors

tylerjw, 2020-01-31:
From https://index.ros.org/doc/ros2/Contributing/Developer-Guide/#exceptions
```Exceptions in destructors should be explicitly avoided```

"Explicitly avoided" to me sounds like this is something that is acceptable sometimes.  How about some version of "Don't throw execptions or call methods that can throw exceptions in destructors".  If this is OK sometimes, please provide an example of when it is acceptable with rational why the example is a good API design.

jacobperron, 2020-01-31:
I don't interpret "explicitly avoided" as "sometimes acceptable". I read it as, "do not let exceptions be thrown from destructors". You can open a PR adding a clarification if you like.

dirk-thomas, 2020-01-31:
While very uncommon you might want to do it if you really know what you are doing and that you are handling it appropriately. Therefore I consider the current wording pretty appropriate.

tfoote, 2020-02-05:
There's a good discussion of it here with examples of where it might be leveraged effectively: https://stackoverflow.com/questions/130117/throwing-exceptions-out-of-a-destructor

tylerjw, 2020-02-06:
After reading that I'm convinced that there might be a few cases where it makes sense but that it is still a generally bad API design as it can result in the stack not unwinding and less useful stack traces.

closed 2020-02-06 by tylerjw