ros2/ros2_documentation #532: change dependency from VS Professional to Visual C++ download

flynneva, 2020-02-17:
the documentation as is now for Windows 10 states that we need Visual Studio to just use ROS2 from binaries.

we should probably change this so its easier for people to get up and running with ROS2 at first....just to the [basic dependency (visual c++ download)](https://support.microsoft.com/en-us/help/2977003/the-latest-supported-visual-c-downloads)

dirk-thomas, 2020-02-18:
Please consider to create a pull request to contribute your suggesting.

flynneva, 2020-02-18:
just submitted the PR

flynneva, 2020-02-18:
looks like this was already included in the eloquent documentation but not back-ported to the dashing documentation....same thing for the python version

tfoote, 2020-02-22:
Fixed in #534

closed 2020-02-22 by tfoote