ros2/ros2_documentation #535: Ambiguous wording...

engband, 2020-02-19:
... concerning ROS2 "systems" vs individual computers running ROS2.

Line 146 of the tutorial at `ros2_documentation/source/Tutorials/Configuring-ROS2-Environment.rst`.

The way "system" is used is ambiguous, and could be interpreted that each individual computer that you want to talk to each other over the ROS network should have a unique ID. Suggestions on how to make it more clear are using input from the following two links, one from discourse, and the other as the 2nd (non-accepted) answer from answers.

I'd be happy to right it myself, but I am unsure as to how much I should expound, or how much expounding is proper in the context of this tutorial.

https://discourse.ros.org/t/how-to-run-example-talker-listener-on-2-machines-with-different-ipaddr/2106/2

https://answers.ros.org/question/301363/how-to-communicate-other-pc-on-ros2/?answer=301760#post-id-301760

maryaB-osr, 2020-02-19:
@engband it'd be great if you opened a PR with your idea of how this can be righted. It seems to me that both your links are saying the same thing as the tutorial, but I'm not against seeing how you'd like to clarify it. As far as how much expounding you should do, I'd say a sentence or two saying something like "A system is..." or "Systems running on the same network that _do_ want to communicate should have the same domain id" (again, not sure which one you're suggesting be expounded on).

Keep in mind this is a beginner tutorial; it's not meant to cover every way or even the best, most thorough way to do something, just to introduce the topic and get the ball rolling without weighing down new users with extraneous details

clalancette, 2020-02-19:
For what it is worth, I do think @engband is right and this can be expounded upon a little bit more here.  But I also agree with @maryaB-osr that we shouldn't go too in-depth here, so rewriting that sentence and/or adding another sentence is probably in order.  @engband please consider opening a PR to reword it, and we can go from there.

engband, 2020-02-19:
Awesome, I'll submit a PR in a few minutes. I just wanted to see if a sentence or a paragraph was proper before submitting it.

engband, 2020-02-19:
Oh, and one other question, is there anywhere in the documentation that says the ROS_DOMAIN_ID needs to be between 0-255? I can mention that, if it isn't. I originally tried to use the MAC address of my computer converted to decimal as the "unique integer" but got an error about it having to be below 255.

maryaB-osr, 2020-02-19:
>  is there anywhere in the documentation that says the ROS_DOMAIN_ID needs to be between 0-255? 

A quick search of ros index says no

I'm going to retract my previous suggestion of "just a sentence" and say add as much as you feel necessary, and we can pare it back afterwards if need be. Just try not to repeat anything from [other places ROS_DOMAIN_ID is mentioned](https://index.ros.org/search/?term=ROS_DOMAIN_ID&section=docs) in the docs (there's not much so you're probably safe)

kscottz, 2020-02-21:
This appears to be merged. I am going to close the issue.

closed 2020-02-21 by kscottz