ros2/ros2_documentation #587: Docs indicate there is an OSX binary installation for ROS2 Dashing, but it's not on github

tleyden, 2020-04-06:
In [these docs](https://github.com/ros2/ros2_documentation/blob/master/source/Installation/Dashing/macOS-Install-Binary.rst#downloading-ros-2) it directs the user to the github releases download page to get the OSX binaries for Dashing, however I'm not seeing that available on the [ROS 2 Dashing Diademata - Patch Release 6](https://github.com/ros2/ros2/releases/tag/release-dashing-20200319) downloads.  

The docs seems a bit misleading here.  Should these docs instead say that binaries for Dashing on OSX are not available, and it should be built from source instead?

clalancette, 2020-04-06:
You can see that we actually did do binary releases for macOS up to Dashing patch release 4.  After that, Apple (and/or Homebrew) dropped support for macOS Sierra (which Dashing was targeting), so we also stopped building binary releases for it.

So yes, a PR clarifying that would be appreciated.

hidmic, 2020-04-08:
@tleyden would you be willing to send a patch explicitly stating what @clalancette described above?

tleyden, 2020-04-08:
@hidmic @clalancette sure, I'll send a PR with that docs update.

clalancette, 2020-04-08:
Thanks, it is appreciated.

commit ce7a2a2c (2020-04-13): Fix #587 Docs indicate there is an OSX binary installation for ROS2 Dashing ... (#605)
closed 2020-04-13 by kscottz