ros2/ros2_documentation #596: Add a documentation section to Contributing guides

maryaB-osr, 2020-04-08:
This should be more of a content-and-structure advisory than rst tips or high level instructions like in the style guide and developer guide. 

Include things like an empty tutorial layout in rst, title conventions, documentation-type definitions (once a distinction between tutorials/guides/concepts/demos is implemented), difficulty-level definitions, etc. 

This can be a place where we accumulate non-technical standards for our documentation (use second person, when to use ` ` ` `, end file names with the file extension e.g. ``example.py``, etc).

pxalcantara, 2021-07-11:
Documentation is very important and also an "easy" way to contribute. How about adding the documentation (REAME) of this repository to the Contributing page in order to help beginners to improve ROS 2 documentation @clalancette? In this way, the documentation could improve faster and reduce the reluctance to migrate to ROS 2.

pxalcantara, 2021-07-18:
> Documentation is very important and also an "easy" way to contribute. How about adding the documentation (REAME) of this repository to the Contributing page in order to help beginners to improve ROS [...]

friendly ping @clalancette

clalancette, 2021-09-24:
> Documentation is very important and also an "easy" way to contribute. How about adding the documentation (REAME) of this repository to the Contributing page in order to help beginners to improve ROS [...]

Rather than copying this README to the Contributing page, I'd rather just link to it from there.  That way we keep the technical information here in this repo, while still providing a way for people to find out how to contribute.  What do you think?

clalancette, 2024-10-11:
We've actually added in a bunch of information to the Contributing page on building documentation now, so I think we can call this resolved and close it.  If you'd like to see more, please open a new, specific issue.

closed 2024-10-11 by clalancette