ros2/ros2_documentation #607: Dead repo link for Foxy Fitzroy

Suyash458, 2020-04-15:
The `ros2.repos` link in the [Windows installation instructions](https://index.ros.org/doc/ros2/Installation/Foxy/Windows-Development-Setup/) for Foxy Fitzroy returns a 404

https://raw.githubusercontent.com/ros2/ros2/foxy/ros2.repos

What should this be updated to?

ahcorde, 2020-04-15:
The freeze of ROS 2 Foxy will [happen soon](https://index.ros.org/doc/ros2/Releases/Release-Foxy-Fitzroy/). The link will be available when the new release is launched.

Suyash458, 2020-04-15:
Oh alright, feel free to close this then

dirk-thomas, 2020-04-15:
Since this branch will only be create close to the release which is more than 1.5 months out the link should be changed to point to master for now.

jacobperron, 2020-04-15:
I've already updated the link for Linux in https://github.com/ros2/ros2_documentation/pull/591

I forgot about the Windows and macOS pages. Here we are: https://github.com/ros2/ros2_documentation/pull/610

closed 2020-04-15 by jacobperron