ros2/ros2_documentation #616: Document copyright requirements for various quality standards

rotu, 2020-04-17:
What are the requirements for maintaining copyright in source files to meet the ROS2 developer guide and quality gates?

In particular, xml, setup.py, package.xml files etc. are all “code” but none seem to have copyrights. Should they?

How does copyright or authorship info need to be maintained? If I edit a file, when must I add my name to the copyright?

The developer guide should clarify these points or state which attribution details are discretionary.

Originally asked (and deemed off topic at):
https://answers.ros.org/question/349677/when-should-i-add-a-copyright-at-the-top-of-a-source-file/

FYI @tfoote, @gavanderhoorn

kscottz, 2020-04-20:
Heads up, this falls under REP-2004. [I cross posted this issue to that open ticket seeking guidance.](https://github.com/ros-infrastructure/rep/pull/218)

wjwwood, 2020-04-21:
I actually don't think the REP is the right place for this information (specifically guidance on which files need a copyright statement and which do not). I've added this to the agenda for the next ROS 2 meeting.

rotu, 2020-04-21:
I think *which files* absolutely belongs in the REP if we say "each source file". XML may or may not be code, but it is unambiguously a source file and no less copyrightable than a C file.

Here's my modest proposal:
1. Add an AUTHORS or CONTRIBUTORS file or put a list in the CONTRIBUTING file. This allows tracking down all contributors if need be.
2. Replace the need for copyright on each file (which is Americentric and unneeded since 1989) with a notice like "This file is subject to copyrights of the authors listed in XXX file and is licensed for use under the terms of the YYY license". This prevents unknowing infringement without introducing a header that needs to be updated each time.
3. The REP has no allowance for distributing files explicitly under *no* copyright. It would be great to formally be able to say in the header "This file is in the public domain". CMakeLists and Launch files benefit greatly from being in the public domain, since they are frequently derived from.
4. Apply this universally to all files or add the expectation that all files not bearing such a header are in the public domain.

clalancette, 2020-04-21:
> Here's my modest proposal:
> 
>     1. Add an AUTHORS or CONTRIBUTORS file or put a list in the CONTRIBUTING file. This allows tracking down all contributors if need be.

This is really unwieldy, and is probably going to be outdated essentially immediately.  The git commits already contain this information; we have used it in the past to track down contributors and change licenses.  I'd suggest just sticking with that.

> 2. Replace the need for copyright on each file (which is Americentric and unneeded since 1989) with a notice like "This file is subject to copyrights of the authors listed in XXX file and is [...]

I'm obviously no lawyer, but it is my understanding that in some jurisdictions, no license != public domain.  I think in some jurisdictions there isn't even a concept of public domain.  I don't want to get into legal discussions here (because they will basically be invalid), but I don't think enshrining this kind of thing in the REP is the right thing to do.

wjwwood, 2020-04-21:
> XML may or may not be code, but it is unambiguously a source file and no less copyrightable than a C file.

I do not think it is as straightforward and unambiguous as you think it is. We don't put copyright in our `package.xml` and we did discuss it (it's not just happenstance). It does need discussion and is not obvious, at least to me.

Further, this starts to bleed into legal territory and what strategy an organization wants to use to secure their copyright, and I do not think we want to be in the position of deciding what is or is not good enough. It's one thing to say "you should have copyright statements" or to say "this is what our projects will do", but it's quite another to tell others what they must do. Some organizations might say "LICENSE" and "COPYRIGHT" file are enough, others may say "literally every file that's not pure docs", and others might be in between (sort of where we are atm), but I think that's a decision for an organization, not a community REP.

I'll let others weigh in like @tfoote and @gerkey perhaps, I don't have the bandwidth right now to decide what should be done here, and do not think it is obvious enough to "just do it".

rotu, 2020-04-21:
It's as much a decision for a REP as "Each source file must have a copyright statement". Maybe we should keep mention of a license in the REP and leave copyright to organizations' discretion just as the choice of license.