ros2/ros2_documentation #626: Anaconda is a better option for Python Installation

songyuc, 2020-04-23:
Hi guys,
I think Anaconda is a better option for Python Installation.

vkuehn, 2020-04-23:
Hi Guy,
please not

songyuc, 2020-04-23:
Hi @vkuehn , why not?

clalancette, 2020-04-23:
Sorry, this isn't really actionable as-is.  To convince us that it is better, you'll need to lay out how it would work for us on all of our Tier-1 platforms (Windows 10, macOS, and Ubuntu), and how it would improve development, deployment, and usage for users.  Also giving some examples on how you think this would be implemented (specific to ROS 2) would be helpful.

songyuc, 2020-04-23:
Hi @clalancette Chris, thanks for your response.
I think there is another substantial reason that I do not quite understand why the ROS2 set the Python path in every script Python file, like
`#!c:\python37\python.exe`
 instead of using a config file which is specific for ROS2 so that, if I had installed a Anaconda Python, then I can set the path in the ROS2-config file like:
`PythonPath  =   C:\Python37\`
If this config file existed, then the issues like https://github.com/ros2/ros2/issues/525 would not happen.

clalancette, 2020-04-23:
I'm not an expert on the Python distribution stuff (particularly on Windows), but I think that's because of using setup.py console_scripts generation to generate the executables that we run.

If you think about it, you need to have *some* default, otherwise it will be broken for everyone out of the box.  On Unix, we just assume `/usr/bin/python3`, and that generally works.  For Windows, we are assuming C:\python37\python.exe, and that works if you follow the installation instructions exactly.

All of that being said, if you have other ideas on how to make this different/better/more configurable, we are open to hearing about them.  Please feel free to open pull requests to the various repositories (probably starting with https://github.com/ros2/ros2cli) showing your ideas.

songyuc, 2020-04-23:
@clalancette Chris, thanks!
I am really interested to realize my idea.
**Setting the Python virtual environment Like PyCharm**
Also, I am wondering whether you are familiar with PyCharm, a Python integrated development environment (IDE), in which developers can specify a Python virtual environment for an individual program.
**Show my idea**
I will try using starting with [https://github.com/ros2/ros2cli](https://github.com/ros2/ros2cli).

kscottz, 2020-04-23:
As this is an opinion more than an actual issue, and it is in the wrong place, I am going to close it.

closed 2020-04-23 by kscottz