ros2/ros2_documentation #628: Copy static content from docs.ros2 here

maryaB-osr, 2020-04-24:
People have been asking about this kind of content a lot. While a better solution for consolidation is being worked on the best we can do is copy the content over here, probably in the Concepts section. The existing content in the Concepts section will have to be checked against the docs.ros2 content to get rid of any overlap.

https://github.com/ros2/docs.ros2.org/issues/17

clalancette, 2020-04-24:
Agreed with this.  Once this is done, I'd also suggest to close all of the open PRs and issues over at https://github.com/ros2/docs.ros2.org and archive the repository.

maryaB-osr, 2020-04-24:
Can we archive it even though the API docs are left behind? Aren't they still relevant/ aren't we going to continue putting each ros2 distro's api docs there?

clalancette, 2020-04-24:
Ah, no, sorry, I didn't realize those were also checked in there.  In that case, when we move the docs over, just make sure to delete them from https://github.com/ros2/docs.ros2.org

maryaB-osr, 2020-04-24:
Now I'm rereading old conversations about moving things around and I feel like it'd be a better idea to move everything from here over there... 

It wouldn't be a huge waste of time to copy that static content over here for now, but... I don't know about archiving docs.ros2. Doesn't it make more sense for all the docs to be on docs.ros2.org than index.ros.org/doc? Not just in terms of this specific issue but in the grand scheme of things.

clalancette, 2021-02-24:
This actually got done now, so closing this out.

closed 2021-02-24 by clalancette