ros2/ros2_documentation #633: [ROS2] Command "ros2 launch turtlesim_mimic_launch.py" Do Not Work as the tutorial said

songyuc, 2020-04-28:
Hi guys,
I am learning about ROS2 on Windows 10 today.
I run the command "ros2 launch turtlesim_mimic_launch.py" as  [3 ros2 launch¶](https://index.ros.org/doc/ros2/Tutorials/Launch-Files/Creating-Launch-Files/#ros2-launch), but the terminal outputs:
> No launch file supplied

and I want to know the reason.
Any idea and answer will be appreciated!

maryaB-osr, 2020-04-28:
Hi @songyuc , `No launch file supplied` means the file is probably not in the directory where you ran the `ros2 launch` command. On Windows CMD you can enter `dir` to see what files are in your current directory; use that to check for `turtlesim_mimic_launch.py`. It could also be possible that the file is there but there's a difference in spelling between the actual file name and the name you wrote after `ros2 launch`.

songyuc, 2020-04-29:
Hi @maryaB-osr , I found "ros2 launch launch/turtlesim_mimic_launch.py" worked.

clalancette, 2020-04-29:
Likely the packages weren't installed properly.  Given this and the other issues you've had, likely you are not following the correct installation and running guide for Windows 10.

But please ask these kinds of questions on https://answers.ros.org.  That way they are all in one place and are easily searchable later. Thanks. Also, it would be better to have one issue rather than 3 different ones.  I'll close this out for now.

songyuc, 2020-04-29:
Hi @clalancette , I cannot agree with you. I think I have carefully followed the tutorial, and there are some commands which are not correct for Windows 10.

maryaB-osr, 2020-04-29:
I just double checked, you're right @songyuc ~ The tutorial never explicitly tells you to go into the `launch` directory to run the command, but it does tell you to put your launch file in there. So yes, trying to launch a file from a different directory will not work. 

> Hi @maryaB-osr , I found "ros2 launch launch/turtlesim_mimic_launch.py" worked.

I'll go add that!

PR #636 (2020-04-29): Creating a launch file patch
fixes #633
closed 2020-04-29 by maryaB-osr