ros2/ros2_documentation #634: [ROS2] There Is Not "sudo" Command on Windows 10 and "rosdep" Do Not Work

songyuc, 2020-04-29:
Hi guys,
I am learning about ROS2 on Windows 10 today.
The ROS2 tutorial told,
[image]

clalancette, 2020-04-29:
Please ask these kinds of questions on https://answers.ros.org. That way they are all in one place and are easily searchable later. Thanks. I'll close this out for now.

songyuc, 2020-04-29:
Hi @clalancette , I do not think this question ought to be asked on [https://answers.ros.org](https://answers.ros.org), because obviously, the command given by the tutorial do not suit Windows 10 system, which need to be corrected.

clalancette, 2020-04-29:
> Hi @clalancette , I do not think this question ought to be asked on https://answers.ros.org, because obviously, the command given by the tutorial do not suit Windows 10 system, which need to be corrected.

The thing is that you need to follow the instructions specific for Windows.  For each of those things where you are having problems on https://index.ros.org/doc/ros2/Tutorials/Turtlesim/Introducing-Turtlesim/, there is a different tab for Linux, macOS, and Windows, and you need to follow the Windows one on Windows 10.  There it does not say anything about `sudo` or `rosdep`, because as you see those do not exist there.

songyuc, 2020-04-29:
> The thing is that you need to follow the instructions specific for Windows. For each of those things where you are having problems on [...]

In fact, the command of "sudo rosdep install -i --from-path src --rosdistro <distro> -y" lies here--[4 Resolve dependencies¶](https://index.ros.org/doc/ros2/Tutorials/Workspace/Creating-A-Workspace/#resolve-dependencies)-- .
Besides, there is not a tab for Windows, and here is the screenshot,
[image]
And I think it will be much better, if there's a tab for Windows.

clalancette, 2020-04-29:
Ah, all right.  Now I see.

In the future, please link to what you are looking at and the exact area you are having problems with.  That will avoid this kind of miscommunication.

@maryaB-osr It looks like we should have some tabs in this page for Linux, macOS, and Windows, as we only run rosdep on the Linux side.

songyuc, 2020-04-29:
@clalancette , thanks sincerely for reopening my issue.
And I will try to describe my issue in a more detailed way.

maryaB-osr, 2020-04-29:
@songyuc For now the tutorials target Linux specifically. See the note at the top of the [tutorials page](https://index.ros.org/doc/ros2/Tutorials/):

> Currently, the beginner level tutorials target Eloquent, installed from Debians on Linux. Certain features and commands may not be available for older distributions/other installation types.

So there's going to be a few things that won't work exactly the same for Windows unfortunately.

PR #637 (2020-04-29): Creating a workspace tutorial patch
fixes #634
closed 2020-04-29 by maryaB-osr