ros2/ros2_documentation #635: [ROS2] "no such command" for using "colcon build --merge-install" on Windows 10

songyuc, 2020-04-29:
Hi guys,
I am learning about ROS2 on Windows 10 today.
I tried using the command "colcon build --merge-install", as the tutorial said
[image]
But the terminal showed "no such command".

Any answer or idea will be appreciated!

clalancette, 2020-04-29:
Please ask these kinds of questions on https://answers.ros.org.  That way they are all in one place and are easily searchable later.  Thanks.  I'll close this out for now.

closed 2020-04-29 by clalancette

songyuc, 2020-04-29:
Hi @clalancette , I have asked this question in [https://answers.ros.org](https://answers.ros.org), but until now, there has been no one who answered my question.

christophebedard, 2020-04-29:
@songyuc you have to wait more than two hours most of the time.

songyuc, 2020-04-29:
@christianrauch , thanks for your tip.
I will check the ROS Answers tomorrow.

dirk-thomas, 2020-04-29:
> I have asked this question in https://answers.ros.org

@songyuc If you make a statement like this then also use a link to point to your question and don't expect readers to go and search for your question.