ros2/ros2_documentation #647: List of supported inter-vendor communication configurations

maryaB-osr, 2020-05-05:
In [Working with multiple RMW implementations](https://index.ros.org/doc/ros2/Tutorials/Working-with-multiple-RMW-implementations/#multiple-rmw-implementations) it says:

> For many cases you will find that nodes using different RMW implementations are able to communicate, however this is not true under all circumstances.
A list of supported inter-vendor communication configurations is forthcoming.

It's been a while since this was written; that list should be added

sloretz, 2020-06-25:
@claireyywang friendly ping :)

Do you plan on working  on this one, or would you like me to re-assign it?

claireyywang, 2020-06-25:
It's on my to-do list, just need to ask and figure it out.

PR #809 (2020-07-08): Add inter-vendor communication configuration
This list is synthesized from https://github.com/ros2/system_tests/blob/master/test_communication/CMakeLists.txt 

closes #647.
closed 2020-07-08 by claireyywang