ros2/ros2_documentation #649: Add a testing tutorial

tfoote, 2020-05-05:
From https://answers.ros.org/question/351474/gtest-for-ros2/

There's a tutorial at: http://wiki.ros.org/gtest which it would be great to have an equivalent for gtest and pytest. Maybe in a combined tutorial with a summary how to run them under colcon

The colcon docs are somewhat limited but should also be referenced and or extended. https://colcon.readthedocs.io/en/released/reference/verb/test.html

kscottz, 2020-05-14:
I've got this RSS tutorial coming up. This would fall under that rubric. I'll see if I can't put something together.

hidmic, 2020-05-26:
@kscottz are you actively pushing this?

kscottz, 2020-05-26:
It should start ramping up next week. I just finished a tranche of events and starting to prep for June/July events.

maryaB-osr, 2020-05-27:
@kscottz I'm happy to take care of this if you don't already have anything going. It fits into my current project. Let me know

kscottz, 2020-05-28:
Go ahead and get started if it is pressing. I probably won't have cycles for this for about a week.

maryaB-osr, 2020-05-29:
@kscottz I don't think it's pressing, and I'm in the same position as you when it comes to timing, so I guess it doesn't make a difference :woman_shrugging: . Let me know if you need anything though!