ros2/ros2_documentation #654: [ROS2] "colon" building turtlesim package failed

songyuc, 2020-05-06:
Hi, guys,
I am learning about ROS2 on Windows 10 today.
The "colon" building turtlesim package failed as the tutorial, [5 Build the workspace with colcon](https://index.ros.org/doc/ros2/Tutorials/Workspace/Creating-A-Workspace/#build-the-workspace-with-colcon).
Specifically, in the step 5 -- [5 Build the workspace with colcon](https://index.ros.org/doc/ros2/Tutorials/Workspace/Creating-A-Workspace/#build-the-workspace-with-colcon), <br> run the command `colcon build --merge-install`,
then an error appeared as,
> Starting >>> turtlesim
> [Processing: turtlesim]
> Failed   <<< turtlesim  [ Exited with code 1 ]
Summary: 0 packages finished [35.5s]
  1 package failed: turtlesim

In order to get the error information, I run the command `colcon build --event-handlers console_cohesion+ --merge-install`, and got the information,
> Starting >>> turtlesim
> [31.534s] colcon.colcon_core.event_reactor ERROR Exception in event handler extension 'console_cohesion': 'utf-8' codec can't decode byte 0xd3 in position 2253: invalid continuation byte
> Traceback (most recent call last):
>   File "c:\python37\lib\site-packages\colcon_core\event_reactor.py", line 78, in _notify_observers
>     retval = observer(event)
>   File "c:\python37\lib\site-packages\colcon_output\event_handler\console_cohesion.py", line 53, in __call__
>     self._lines[job]).decode() + \
> UnicodeDecodeError: 'utf-8' codec can't decode byte 0xd3 in position 2253: invalid continuation byte
> 
> Failed   <<< turtlesim  [ Exited with code 1 ]
> 
> Summary: 0 packages finished [29.4s]
>   1 package failed: turtlesim

Could you please tell me how to fix this?
You answer and idea will be appreciated!

maryaB-osr, 2020-05-07:
hi @songyuc, this type of question should be asked on https://answers.ros.org/questions/
the issues here on ros2_documentation are meant to fix parts of the documentation, not for debugging.
once you get an answer on answers.ros.org, if you feel some changes need to be made to the documentation, come back and open another issue.

closed 2020-05-07 by maryaB-osr

songyuc, 2020-05-08:
Hi @maryaB-osr !
I asked the same kind of question, but it hasn't been solved yet.
Could you please help me solve this issue, ["[ROS2] "colon" building turtlesim package failed"](https://answers.ros.org/question/351567/ros2-colon-building-turtlesim-package-failed/)?

maryaB-osr, 2020-05-11:
Were you able to work through that problem? It seems like, according to this PR, the issue might be behind the use of `'` instead of `"` for Windows.