ros2/ros2_documentation #695: Maybe missing step for installing colon.

abrzozowski, 2020-05-16:
Hi there, I'm not sure I maybe missed some chapter.

During my first step into **ROS2**, I've been followed sub-pages from https://index.ros.org/doc/ros2

When I completed _Installation_ pages with the _ROS 2 Eloquent Elusor_ by _Linux (Ubuntu Bionic(18.04))_ -> **Debian packages** and _Beginner: CLI Tools_ tutorials, I've begun **Beginner: Client Libraries** tutorial. 

On the **5 Build the workspace with colcon** step I didn't have installed **colcon** command 
`colcon: command not found`

Maybe the following link for a colon should be added:
https://index.ros.org/doc/ros2/Installation/Eloquent/Linux-Development-Setup/#install-development-tools-and-ros-tools
or
https://index.ros.org/doc/ros2/Tutorials/Colcon-Tutorial?

maryaB-osr, 2020-05-18:
Hi @abrzozowski, I'm not sure why colcon could not be found; since you installed from Debian packages, it should be included. Anyway, I'll add a link to the colcon tutorial at that step just in case anyone else has the same problem. Thanks for letting us know

clalancette, 2020-05-19:
> Hi @abrzozowski, I'm not sure why colcon could not be found; since you installed from Debian packages, it should be included.

It is somewhat by design that it is not included.  It may be the case that you just want to deploy the debians on a robot and not build anything with colcon, in which case the dependency would be spurious.  It does make initial development harder, though.

> Anyway, I'll add a link to the colcon tutorial at that step just in case anyone else has the same problem.

For what it is worth, there is already a link to installing colcon at the top of that tutorial, so I don't think we need to do much more here.

clalancette, 2020-05-19:
@maryaB-osr I leave it to you whether to close this out or not.

maryaB-osr, 2020-05-19:
> there is already a link to installing colcon at the top of that tutorial

ah ok, missed that. i'll close this then

closed 2020-05-19 by maryaB-osr

abrzozowski, 2020-05-20:
I missed colcon installation from **prerequisites** too, sorry for that.