ros2/ros2_documentation #712: Missing ros2/example_interfaces as prerequisite in 'Using colcon to build packages' chapter

crystaldust, 2020-05-24:
Hi, I've followed the tutorial [Using colcon to build packages](https://index.ros.org/doc/ros2/Tutorials/Colcon-Tutorial/) and got the error complaining:
```bash
Starting >>> example_interfaces
Starting >>> examples_rclcpp_minimal_composition
Starting >>> examples_rclcpp_minimal_publisher
Starting >>> examples_rclcpp_minimal_subscriber
--- stderr: example_interfaces                                                                                             
CMake Error: The source directory "/home/lance/ros2_example_ws/src/example_interfaces" does not exist.
Specify --help for usage, or press the help button on the CMake GUI.
make: *** [cmake_check_build_system] Error 1
---
Failed   <<< example_interfaces	[ Exited with code 2 ]
Aborted  <<< examples_rclcpp_minimal_publisher                                                                                                           
Aborted  <<< examples_rclcpp_minimal_subscriber
Aborted  <<< examples_rclcpp_minimal_composition                                         

Summary: 0 packages finished [0.78s]
  1 package failed: example_interfaces
  3 packages aborted: examples_rclcpp_minimal_composition examples_rclcpp_minimal_publisher examples_rclcpp_minimal_subscriber
  1 package had stderr output: example_interfaces
  13 packages not processed
```

The error was mentioned in a ROS Answer thread [ROS2 colcon build fails on demo pkg docker](https://answers.ros.org/question/331633/ros2-colcon-build-fails-on-demo-pkg-docker/), so after putting the [example_interfaces](https://github.com/ros2/example_interfaces) repo to `src/` folder, everything goes right.

Will this worth a documentation update, putting [example_interfaces](https://github.com/ros2/example_interfaces) as a prerequisite to make the build smooth? Since I'm pretty new to ROS(2), not sure if this is any context that I miss.

clalancette, 2020-05-26:
Typically, we are assuming that `example_interfaces` would come from the underlay (i.e. `/opt/ros/eloquent`).  And in a default install (`sudo apt-get install ros-eloquent-desktop`), that should come in automatically.  How did you install ROS 2?

crystaldust, 2020-05-26:
@clalancette Aha, that should be my problem, I installed ros-dashing-ros-base(the bare bone package) so example_interfaces is missing in the underlay.

By the way, I've been recently wondering if there is a 'package management mechanism' for ROS2. Just like golang or Python, we can specifiy a ROS(2) pkg hosted on github or a index registry, and install it just by some command like 'ros2 install github.com/ros2/example_interfaces'. Since I'm new to ROS(2), not sure if there it is already there(all the search result of 'ros2 package install' is about installing ros2 itself and related packages)

clalancette, 2020-05-26:
> By the way, I've been recently wondering if there is a 'package management mechanism' for ROS2.

Instead of having a custom one, we are reusing the system package installation (apt-get on Debian/Ubuntu, dnf on Fedora).  When package maintainers want to release their packages to the community, they open a PR to https://github.com/ros/rosdistro/, and then our buildfarm picks that up and builds packages for them.  This currently only works for Linux-based distributions, but there is no reason in principle why we couldn't extend it to use homebrew on macOS and one of the package managers on Windows.  It's just a bunch of work to get there and there hasn't been a lot of call for it.

> @clalancette Aha, that should be my problem, I installed ros-dashing-ros-base(the bare bone package) so example_interfaces is missing in the underlay.

FYI, there is a note at the top of that article that you need to install the `desktop` variant for the tutorial to work.  So with that, I'm actually going to close this out.  Feel free to reopen if you think there is something else that needs to be done here.

closed 2020-05-26 by clalancette

crystaldust, 2020-05-27:
Thanks, Chris. That's enough info. Btw, I'm recently looking into the ros2cli source code and see there is a plugin framework to extend the command-line tool. So it should be possible to make the source code package install by extension. I'll make a quick implementation and see if it helps the developers. It would make the manual source build easier for developers who don't use Debian/RPM based distributions.