ros2/ros2_documentation #722: Topic Statistics information

emersonknapp, 2020-05-30:
New topic statistics feature doesn't say how to use it. Concepts, Tutorial, or both needed.

sloretz, 2020-06-11:
@emersonknapp I labeled this one as Help Wanted for now. Would you be willing to contribute a Topic Statistics tutorial?

emersonknapp, 2020-06-12:
Someone on our team will be providing this documentation within the next week - you can assign me if you like and I will follow up on making sure this issue is tied to that contribution.

dabonnie, 2020-06-12:
FYI this is also tracked in the ROS2 meta ticket (https://github.com/ros2/ros2/issues/917), where the [demo](https://github.com/ros-tooling/aws-roadmap/issues/262) and the [concepts wiki entry](https://github.com/ros-tooling/aws-roadmap/issues/245) are both on the [aws-roadmap](https://github.com/ros-tooling/aws-roadmap/issues) and in progress.

dabonnie, 2020-06-16:
> @emersonknapp I labeled this one as Help Wanted for now. Would you be willing to contribute a Topic Statistics tutorial?

Please see #767

PR #767 (2020-06-19): Add Topic Statistics to concepts
Related to https://github.com/ros2/ros2/issues/917
Moves https://github.com/ros-tooling/aws-roadmap/issues/245
Resolves #722
closed 2020-06-19 by maryaB-osr