ros2/ros2_documentation #756: Add documentation of the scope of binary packaging jobs

tfoote, 2020-06-08:
The install from Binary instructions such as https://index.ros.org/doc/ros2/Installation/Foxy/#binary-packages does not mention that the binary packages for windows does not include all packages in Foxy.

When this was first written all packages were in the repos files, but now there are other packages that are not included in the binary packages, so we should document what the scope of packages is that are packaged.

As a related discussion right now we have only one set of packaging jobs. However we have several clearly defined entry points that we're managing https://github.com/ros2/variants in REP 2001 https://ros.org/reps/rep-2001.html And we package docker images of those variants. it might make sense for us to setup our packaging jobs to match those instead of documenting a 3rd non-standard variant.

We should also document how to add more of the released packages to your overlay workspace using tools like rosinstall_generator.

tfoote, 2020-06-09:
We discussed this in the ROS team meeting.

Since Foxy and earlier are already out we will add documentation for the existing ros2.repos scope.

Long term we would like to move away from the ros2.repos file and to that end in the upcoming rolling release we will plan to migrate to use a variant based replacement. 

We will either:
- Create a new variant to replicate the current ros2.repos scope
- Confirm that an existing variant is appropriate, or adjust it to match.

jacobperron, 2020-06-29:
See https://github.com/ros2/ros2_documentation/pull/802, which adds a note to the top of installation instructions involving fat archives setting expectations of what packages will be installed.

commit e77e6b1c (2020-06-29): Document fat binary installations do not contain all ROS 2 packages

Resolves #756.

Add a note to the top of all pre-built binary package installation instructions.
The note describes the package list with respect to the variants in REP 2001 and links to the appropriate ros2.repos file.
PR #802 (2020-06-29): Document fat binary installations do not contain all ROS 2 packages
Resolves #756.

Add a note to the top of all pre-built binary package installation instructions.
The note describes the package list with respect to the variants in REP 2001 and links to the appropriate ros2.repos file.
commit 1af29bd3 (2020-06-29): Document fat binary installations do not contain all ROS 2 packages (#802)

Resolves #756.

Add a note to the top of all pre-built binary package installation instructions.
The note describes the package list with respect to the variants in REP 2001 and links to the appropriate ros2.repos file.
closed 2020-06-29 by jacobperron