ros2/ros2_documentation #759: "ADLINK OpenSplice Debian Packages built by OSRF" not available for ROS2 Foxy

hazevedosa, 2020-06-11:
Running Foxy in ubuntu 20.04, the outcome of 'sudo apt install libopensplice69' is:

```
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package libopensplice69
```

clalancette, 2020-06-11:
As of Foxy, opensplice is no longer a supported DDS backend.  You can see which DDS vendors we support on https://ros.org/reps/rep-2000.html#id32 .

Where did you see the instructions to install opensplice?

christophebedard, 2020-06-11:
This section: https://index.ros.org/doc/ros2/Installation/Foxy/Linux-Development-Setup/#install-additional-dds-implementations-optional

links to this (which doesn't seem to be distro-specific): https://index.ros.org/doc/ros2/Installation/DDS-Implementations/

hazevedosa, 2020-06-11:
@christophebedard is correct... I was trying to get rid of the "[connext_cmake_module] Warning: (...)" thing when I source ROS2. Should we add a note to the doc?

clalancette, 2020-06-11:
I see.  I've opened up https://github.com/ros2/ros2_documentation/pull/760 to clarify that OpenSplice is only supported on Eloquent and older.

clalancette, 2020-06-11:
> @christophebedard is correct... I was trying to get rid of the "[connext_cmake_module] Warning: (...)" thing when I source ROS2. Should we add a note to the doc?

Just to be clear; Connext has nothing at all to do with OpenSplice.  They are two separate DDS implementations.  What I did in #760 should take care of the OpenSplice portion.

Installing the Connext binaries should indeed get rid of that warning.

closed 2020-06-11 by clalancette

hazevedosa, 2020-06-11:
great, thanks!

PR #760 (2020-06-11): Tabify DDS-Implementations.rst.
Since we don't support OpenSplice on Foxy and newer,
tabify this page to explain that.

Fixes #759