ros2/ros2_documentation #765: Add different instructions for sourcing ROS 2 in (Foxy / Ubuntu 20.04 and newer)

hazevedosa, 2020-06-12:
`source /opt/ros/foxy/setup.bash` does not work;
`. /ros2_foxy/ros2-linux/setup.bash` works;

hazevedosa, 2020-06-12:
The generated files are different:

1. This is `/ros2_foxy/ros2-linux/setup.bash`:

```
# generated from colcon_bash/shell/template/prefix_chain.bash.em

# This script extends the environment with the environment of other prefix
# paths which were sourced when this file was generated as well as all packages
# contained in this prefix path.

# function to source another script with conditional trace output
# first argument: the path of the script
_colcon_prefix_chain_bash_source_script() {
  if [ -f "$1" ]; then
    if [ -n "$COLCON_TRACE" ]; then
      echo ". \"$1\""
    fi
    . "$1"
  else
    echo "not found: \"$1\"" 1>&2
  fi
}

# source this prefix
# setting COLCON_CURRENT_PREFIX avoids determining the prefix in the sourced script
COLCON_CURRENT_PREFIX="$(builtin cd "`dirname "${BASH_SOURCE[0]}"`" > /dev/null && pwd)"
_colcon_prefix_chain_bash_source_script "$COLCON_CURRENT_PREFIX/local_setup.bash"

unset COLCON_CURRENT_PREFIX
unset _colcon_prefix_chain_bash_source_script

```

This is `/opt/ros/foxy/setup.bash`:

```
# copied from ament_package/template/prefix_level/setup.bash

AMENT_SHELL=bash

# source setup.sh from same directory as this file
AMENT_CURRENT_PREFIX=$(builtin cd "`dirname "${BASH_SOURCE[0]}"`" && pwd)
# trace output
if [ -n "$AMENT_TRACE_SETUP_FILES" ]; then
  echo ". \"$AMENT_CURRENT_PREFIX/setup.sh\""
fi
. "$AMENT_CURRENT_PREFIX/setup.sh"
```

clalancette, 2020-06-12:
Sorry, it's not clear what is not working for you.  Can you explain more fully what you are trying to do, and what is not working for you?

hazevedosa, 2020-06-12:
Sorry if I wasn't clear.
When I try to source `/opt/ros/foxy/setup.bash`, ros2 is not recognized:

```
$ source /opt/ros/foxy/setup.bash
$ ros2
ros2: command not found

```

When I source `/ros2_foxy/ros2-linux/setup.bash`, it is recognized:

```
$ source ~/ros2_foxy/ros2-linux/setup.bash
$ ros2
usage: ros2 [-h] Call `ros2 <command> -h` for more detailed usage. ...

```

hazevedosa, 2020-06-12:
And... in the tutorial page, the instruction is to source with:

`source /opt/ros/<distro>/setup.bash`

clalancette, 2020-06-12:
OK, I see.  Most likely you don't have the correct packages installed.  If you run:

`$ dpkg -l | grep foxy-desktop`

what do you get?  Also, if you run:

```
$ dpkg -l | grep foxy-ros2cli
```

what do you get?

hazevedosa, 2020-06-12:
I get nothing...

clalancette, 2020-06-12:
> I get nothing... If I don't have the correct packages, there might be something missing in the installation tutorial for foxy. I followed it step-by-step.

https://index.ros.org/doc/ros2/Installation/Foxy/Linux-Install-Debians/#id4 does recommend to install `ros-foxy-desktop`.  Which instructions did you follow?

hazevedosa, 2020-06-12:
This one (that does not mention `ros-foxy-desktop`)
https://index.ros.org/doc/ros2/Installation/Foxy/Linux-Install-Binary/

clalancette, 2020-06-12:
> This one ("fat" archive --- which does not mention `ros-foxy-desktop`):
> https://index.ros.org/doc/ros2/Installation/Foxy/Linux-Install-Binary/

Right, so for the Linux Binary install, it is all included.  So that's why it works fine for you when you source the install binary.

At this point, it looks like this is a configuration issue.  I'm going to go ahead and close this; if you have further questions on this topic, I'll suggest opening a question on answers.ros.org.

closed 2020-06-12 by clalancette