ros2/ros2_documentation #768: Launch file error in tutorial, dashing

Met7, 2020-06-17:
When running the launch file in the tutorial 
https://index.ros.org/doc/ros2/Tutorials/Using-Parameters-In-A-Class-CPP/#change-via-a-launch-file
on dashing, it fails with 
[ERROR] [launch]: Caught exception in launch (see debug for traceback): __init__() got an unexpected keyword argument 'emulate_tty'

The parameter is indeed necessary. Without it it launches, but the console is not updating - only flushes all the output on exit.

- Operating System: Ubuntu 18.04.4 LTS
- Installation type: binaries
- Version or commit hash: dashing
- DDS implementation: default
- Client library (if applicable): cpp

maryaB-osr, 2020-06-19:
I was able to reproduce this error. @wjwwood I think you worked on `emulate_tty` on Dashing/before Eloquent came out; is it supposed to work in Dashing?

wjwwood, 2020-06-20:
It was added in version `0.9.0`:

https://github.com/ros2/launch/blob/master/launch/CHANGELOG.rst#090-2019-09-18

But in Dashing the `launch` version is `0.8.7`:

https://index.ros.org/p/launch/github-ros2-launch/#dashing

So, unfortunately, no that option isn't in dashing.

maryaB-osr, 2020-06-23:
Accidentally committed this fix straight to master :(
cfe3672 

Going to close this since it's done

closed 2020-06-23 by maryaB-osr