ros2/ros2_documentation #774: Add documentation for Rolling Ridley

nuclearsandwich, 2020-06-18:
Tracking issue to include updates for Rolling Ridley

- [ ] Entry on [Releases page](https://index.ros.org/doc/ros2/Releases/)
- [ ] Release page specifically for Rolling Ridley e.g. https://index.ros.org/doc/ros2/Releases/Release-Rolling-Ridley
- [ ] Installation guide for Rolling Ridley e.g. https://index.ros.org/doc/ros2/Installation/Rolling
- [ ] Mention on [Installation page](https://index.ros.org/doc/ros2/Installation/) I don't necessarily thing it should get an entry in the table along side the supported releases but a titled paragraph below explaining the target audience for the Rolling distribution and a link to the rolling installation instructions seems appropriate.
- [ ] Update [Releasing a ROS 2 package with bloom](https://index.ros.org/doc/ros2/Tutorials/Releasing-a-ROS-2-package-with-bloom/) with information on how the Rolling release will interact with the stable release process.

maryaB-osr, 2020-06-19:
@nuclearsandwich What about [this page](https://index.ros.org/doc/ros2/Installation/Latest-Development-Setup/) and its corresponding [section on the installation homepage](https://index.ros.org/doc/ros2/Installation/#contributing-to-ros-2-core)? I think those are kind of the same thing as the rolling release

nuclearsandwich, 2020-06-19:
> Would rewording that section on the installation page satisfy the fourth task listed above?

Yes! I think @tfoote suggested this in a hangout we just had. Clearly I ought've invited you too!

> What about [this page](https://index.ros.org/doc/ros2/Installation/Latest-Development-Setup/) ... I think those are kind of the same thing as the rolling release

#776 updates the `$PLATFORM-latest` labels used on that page to the Rolling instructions instead of the latest stable distribution. There will always be some changes on development branches that haven't yet been released into the Rolling distribution so it isn't quite one-and-the-same with the Latest Development article. Aside from those links, I don't see a need to mention the Rolling distribution explicitly, but I could be overlooking something.

For the Rolling installation instructions in #776 I link *from* the `Install-Binary` pages *to* the [Prerelease testing page](https://index.ros.org/doc/ros2/Installation/Prerelease-Testing/) but not the other way around. The links on the prerelease testing page take you to the installation homepage since it's also possible to do pre-release testing for releases of stable distributions.

chapulina, 2020-09-23:
Is this done?

nuclearsandwich, 2020-10-09:
Yes!

closed 2020-10-09 by nuclearsandwich