ros2/ros2_documentation #781: Explicitly state that there should be a space between "ROS" and "2"

christophebedard, 2020-06-20:
Note: this is mostly a suggestion. If you don't think that's necessary, then that's fine with me.

It is generally accepted that the "proper" use is "ROS 2" with a space in between. However, as far as I can see, there is nothing officially stating that (other than this documentation using "ROS 2" pretty much everywhere). This would be useful, even if it's only to be able to point people to it as a "source," hopefully without being too anal about it.

Thus it would be nice to explicitly write that down somewhere here in the docs. I could do it, but I'm just wondering where it should go exactly. Perhaps on the landing page, i.e. `source/index.rst`? Also this could simply be adding a quick note pointing out the space.

tfoote, 2020-06-21:
@kscottz Have we posted the new logo guidelines yet?

christophebedard, 2020-06-22:
I did find this: https://www.ros.org/press-kit/

> Specific versions of ROS should be specified by placing the release name or version number after the ROS name, separated by a space. For example, ROS Melodic, ROS Dashing, ROS 1, ROS 2.

kscottz, 2020-07-20:
ROS branding guidelines are going to end up being a secondary repository for a variety of reasons. We actually have the new branding guide finished but we're holding off publication of it until we finish the Gazebo guide. What I can do is create a link to the repository where we intend to keep this information.

christophebedard, 2020-07-20:
Sounds good!

closed 2020-07-21 by kscottz