ros2/ros2_documentation #797: problems when creating custom ROS 2 msg and srv files

shanpenghui, 2020-06-28:
PROBLEM ONE:
First, i compile with the problems below:
```
sph@sph-HP-ZBook-14-G2:~/Documents/ros2_ws$ colcon build --packages-select my_pkg
Starting >>> my_pkg  
[Processing: my_pkg]                             
--- stderr: my_pkg                                
CMake Warning (dev) in /home/sph/Documents/ros2_ws/build/my_pkg/connext_cmake_module/check_abi/CMakeLists.txt:
  No project() command is present.  The top-level CMakeLists.txt file must
  contain a literal, direct call to the project() command.  Add a line of
  code such as

    project(ProjectName)

  near the top of the file, but after cmake_minimum_required().

  CMake is pretending there is a "project(Project)" command on the first
  line.
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at /usr/local/share/cmake-3.17/Modules/FindPackageHandleStandardArgs.cmake:272 (message):
  The package name passed to `find_package_handle_standard_args` (OpenSplice)
  does not match the name of the calling package (opensplice).  This can lead
  to problems in calling code that expects `find_package` result variables
  (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  /usr/share/opensplice/cmake/opensplice-config.cmake:73 (find_package_handle_standard_args)
  /opt/ros/dashing/share/opensplice_cmake_module/cmake/Modules/FindOpenSplice.cmake:100 (find_package)
  /opt/ros/dashing/share/rosidl_typesupport_opensplice_c/cmake/rosidl_typesupport_opensplice_c-extras.cmake:6 (find_package)
  /opt/ros/dashing/share/rosidl_typesupport_opensplice_c/cmake/rosidl_typesupport_opensplice_cConfig.cmake:38 (include)
  /opt/ros/dashing/share/rosidl_default_generators/cmake/rosidl_default_generators-extras.cmake:21 (find_package)
  /opt/ros/dashing/share/rosidl_default_generators/cmake/rosidl_default_generatorsConfig.cmake:38 (include)
  CMakeLists.txt:24 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

---
Finished <<< my_pkg [48.3s]

Summary: 1 package finished [48.5s]
  1 package had stderr output: my_pkg

```
After that, I change nothing! nothing!
Comes the result:

```
sph@sph-HP-ZBook-14-G2:~/Documents/ros2_ws$ colcon build --packages-select my_pkg
Starting >>> my_pkg  
Finished <<< my_pkg [0.86s]                     

Summary: 1 package finished [1.05s]

```

I get confused. Somebody can tell what happen? And why?

Code is here:
https://github.com/shanpenghui/ros2_pkgs_.git

clalancette, 2020-06-28:
When you open issues like this, please provide a lot of information about your environment.  That includes the operating system you are using, the ROS distribution, and anything out of the ordinary you may have installed.  Also include step-by-step instructions on what you have done.

In this case, I tried your test repository on ROS Dashing, Eloquent, and Foxy, and it compiled on all 3 just fine.  So I'm guessing there is something wrong with your environment, but we'll need more information to know for sure.

shanpenghui, 2020-07-03:
> When you open issues like this, please provide a lot of information about your environment. That includes the operating system you are using, the ROS distribution, and anything out of the ordinary [...]

Tks for replying.
**Computer Details**

> OS and Version: Ubuntu, 18.04
> ROS distribution: dashing(installed by debian packages)
> Compiler Version: G++-7

The steps to compile is very simple.Just as below:
1.git clone https://github.com/shanpenghui/vscode_ros2_ws.git 
2.cd vscode_ros2_ws
3.source /opt/ros/dashing/setup.bash
4.colcon build --packages-select tutorial_interfaces
5.get the results

Here is my ~/.bashrc file
```
# ~/.bashrc: executed by bash(1) for non-login shells.
# see /usr/share/doc/bash/examples/startup-files (in the package bash-doc)
# for examples

# If not running interactively, don't do anything
case $- in
    *i*) ;;
      *) return;;
esac

# don't put duplicate lines or lines starting with space in the history.
# See bash(1) for more options
HISTCONTROL=ignoreboth

# append to the history file, don't overwrite it
shopt -s histappend

# for setting history length see HISTSIZE and HISTFILESIZE in bash(1)
HISTSIZE=1000
HISTFILESIZE=2000

# check the window size after each command and, if necessary,
# update the values of LINES and COLUMNS.
shopt -s checkwinsize

# If set, the pattern "**" used in a pathname expansion context will
# match all files and zero or more directories and subdirectories.
#shopt -s globstar

# make less more friendly for non-text input files, see lesspipe(1)
[ -x /usr/bin/lesspipe ] && eval "$(SHELL=/bin/sh lesspipe)"

# set variable identifying the chroot you work in (used in the prompt below)
if [ -z "${debian_chroot:-}" ] && [ -r /etc/debian_chroot ]; then
    debian_chroot=$(cat /etc/debian_chroot)
fi

# set a fancy prompt (non-color, unless we know we "want" color)
case "$TERM" in
    xterm-color|*-256color) color_prompt=yes;;
esac

# uncomment for a colored prompt, if the terminal has the capability; turned
# off by default to not distract the user: the focus in a terminal window
# should be on the output of commands, not on the prompt
#force_color_prompt=yes

if [ -n "$force_color_prompt" ]; then
    if [ -x /usr/bin/tput ] && tput setaf 1 >&/dev/null; then
	# We have color support; assume it's compliant with Ecma-48
	# (ISO/IEC-6429). (Lack of such support is extremely rare, and such
	# a case would tend to support setf rather than setaf.)
	color_prompt=yes
    else
	color_prompt=
    fi
fi

if [ "$color_prompt" = yes ]; then
    PS1='${debian_chroot:+($debian_chroot)}\[\033[01;32m\]\u@\h\[\033[00m\]:\[\033[01;34m\]\w\[\033[00m\]\$ '
else
    PS1='${debian_chroot:+($debian_chroot)}\u@\h:\w\$ '
fi
unset color_prompt force_color_prompt

# If this is an xterm set the title to user@host:dir
case "$TERM" in
xterm*|rxvt*)
    PS1="\[\e]0;${debian_chroot:+($debian_chroot)}\u@\h: \w\a\]$PS1"
    ;;
*)
    ;;
esac

# enable color support of ls and also add handy aliases
if [ -x /usr/bin/dircolors ]; then
    test -r ~/.dircolors && eval "$(dircolors -b ~/.dircolors)" || eval "$(dircolors -b)"
    alias ls='ls --color=auto'
    #alias dir='dir --color=auto'
    #alias vdir='vdir --color=auto'

    alias grep='grep --color=auto'
    alias fgrep='fgrep --color=auto'
    alias egrep='egrep --color=auto'
fi

# colored GCC warnings and errors
#export GCC_COLORS='error=01;31:warning=01;35:note=01;36:caret=01;32:locus=01:quote=01'

# some more ls aliases
alias ll='ls -alF'
alias la='ls -A'
alias l='ls -CF'

# Add an "alert" alias for long running commands.  Use like so:
#   sleep 10; alert
alias alert='notify-send --urgency=low -i "$([ $? = 0 ] && echo terminal || echo error)" "$(history|tail -n1|sed -e '\''s/^\s*[0-9]\+\s*//;s/[;&|]\s*alert$//'\'')"'

# Alias definitions.
# You may want to put all your additions into a separate file like
# ~/.bash_aliases, instead of adding them here directly.
# See /usr/share/doc/bash-doc/examples in the bash-doc package.

if [ -f ~/.bash_aliases ]; then
    . ~/.bash_aliases
fi

# enable programmable completion features (you don't need to enable
# this, if it's already enabled in /etc/bash.bashrc and /etc/profile
# sources /etc/bash.bashrc).
if ! shopt -oq posix; then
  if [ -f /usr/share/bash-completion/bash_completion ]; then
    . /usr/share/bash-completion/bash_completion
  elif [ -f /etc/bash_completion ]; then
    . /etc/bash_completion
  fi
fi

#source /opt/ros/melodic/setup.bash
#export ROS_PACKAGE_PATH=${ROS_PACKAGE_PATH}:/home/sph/Documents/ORB_SLAM
```

The problem is still there.

clalancette, 2020-07-06:
There are two things here:

1.  You are using a custom version of cmake (3.17), not the system version.  That's probably the cause of the warnings.
1.  The warnings are not fatal; as you can see at the end, it still compiles.

My suggestion is to uninstall the custom version of cmake that you are using, and just use the system version.  If that is not an option for some reason, then you can continue to use cmake 3.17, you'll just have to put up with the warnings.

I'm going to go ahead and close this for now, but feel free to keep commenting or reopen if you think this was closed in error.

closed 2020-07-06 by clalancette