ros2/ros2_documentation #806: ros2 doctor error : invalid choice: 'doctor'

shanpenghui, 2020-07-03:
The error comes out when I try to use the command
```
ros2 doctor
```
The screen output is :
```
usage: ros2 [-h] Call `ros2 <command> -h` for more detailed usage. ...
ros2: error: argument Call `ros2 <command> -h` for more detailed usage.: invalid choice: 'doctor' (choose from 'action', 'component', 'daemon', 'extension_points', 'extensions', 'launch', 'lifecycle', 'msg', 'multicast', 'node', 'param', 'pkg', 'run', 'security', 'service', 'srv', 'topic')
```
Im sure to have been installed everything about ros-dashing. What is the reason?

christophebedard, 2020-07-03:
`ros2doctor` was only added in Eloquent: https://index.ros.org/doc/ros2/Releases/Release-Eloquent-Elusor/#new-features-in-this-ros-2-release

Were you following this tutorial? https://index.ros.org/doc/ros2/Tutorials/Getting-Started-With-Ros2doctor/ Perhaps the tutorial should mention that it's only available on Eloquent and newer.

clalancette, 2020-07-06:
Yes, @christophebedard is exactly right, `ros2 doctor` is only available in Eloquent and later.

@maryaB-osr Would you mind doing an update to this tutorial to mention that it is for Eloquent and later only?

PR #812 (2020-07-09): Add note that ros2doctor isn't in Dashing
fixes #806 

(also fixed a small unrelated syntax warning)
closed 2020-07-09 by maryaB-osr