ros2/ros2_documentation #807: Switch to setup.sh instead of setup.bash

DamianRivas, 2020-07-03:
When installing ROS2 I ran into issues sourcing `setup.bash` (probably because I use zsh?).

Looking inside `setup.bash`, all it does is source `setup.sh` anyway, so why not tell users to do `source /opt/ros/<distro>/setup.sh` instead?

https://index.ros.org/doc/ros2/Tutorials/Configuring-ROS2-Environment/

christophebedard, 2020-07-03:
> When installing ROS2 I ran into issues sourcing setup.bash (probably because I use zsh?).

If you've installed `colcon-zsh` (you should have it if you've installed `colcon-common-extensions`, but you can check with `colcon version-check`), you should also see a `setup.zsh` file, which is meant for zsh.

DamianRivas, 2020-07-03:
Personally I am using setup.zsh, but that wasn't why I opened this issue.

Wouldn't it be better for the docs to say source setup.sh since that's all that setup.bash is doing?

christophebedard, 2020-07-03:
Ah, sorry!

I don't really see where setup.bash just calls setup.sh. It's probably bash by default because that's the default shell on Ubuntu, this way the tutorials won't need to be changed from sh back to bash if something breaks (even if bash and sh are mostly compatible). But I'd let the right people properly answer this one.

clalancette, 2020-07-06:
> Wouldn't it be better for the docs to say source setup.sh since that's all that setup.bash is doing?

I don't think so.  Technically `bash` and `sh` are different things, but `bash` is a strict superset.  While it is true that at the moment they do almost the same things, I can imagine a case where we decide to use a bash feature that is not available in sh.

I'm going to go ahead and close this, but feel free to keep commenting or reopen if you think it was closed in error.

closed 2020-07-06 by clalancette

dirk-thomas, 2020-07-06:
> Looking inside `setup.bash`, all it does is source `setup.sh` anyway

When sourcing the `.bash` and `.zsh` version of the setup file it already does perform shell specific things. The most important one is setting up completion for these shells.

While it may not be obvious when looking at the `setup.bash` file it also set the `AMENT_SHELL` to `bash` which will down the road cause `bash`-specific behavior to be triggered.

DamianRivas, 2020-07-06:
> it also set the `AMENT_SHELL` to `bash` which will down the road cause `bash`-specific behavior to be triggered

Ah okay! I took a much closer look and did notice the `sh` script checks that variable to determine which shell it's in before branching to different behaviours. This is good to know. Thank you!