ros2/ros2_documentation #808: rqt Issue on Windows 10

MicheleGalliano, 2020-07-06:
Hi, when I use rqt thi is the output:
```

C:\WINDOWS\system32>rqt
RosPluginProvider._parse_plugin_xml() plugin file "D:\Programmi\ROS\ros2-windows\share\qt_gui_cpp/plugin.xml" in package "qt_gui_cpp" not found
RosPluginProvider._parse_plugin_xml() plugin file "D:\Programmi\ROS\ros2-windows\share\rqt_gui_cpp/plugin.xml" in package "rqt_gui_cpp" not found
RosPluginProvider._parse_plugin_xml() plugin file "D:\Programmi\ROS\ros2-windows\share\qt_gui_cpp/plugin.xml" in package "qt_gui_cpp" not found
RosPluginProvider._parse_plugin_xml() plugin file "D:\Programmi\ROS\ros2-windows\share\rqt_gui_cpp/plugin.xml" in package "rqt_gui_cpp" not found
[ERROR] [1594050596.237395900] [get_service_class]: Malformed srv message_type: turtlesim/action/RotateAbsolute_GetResult
ServiceCaller.on_refresh_services_button_clicked(): could not get class turtlesim/action/RotateAbsolute_GetResult of service /turtle1/rotate_absolute/_action/get_result. Skipping it.
[ERROR] [1594050596.242802200] [get_service_class]: Malformed srv message_type: turtlesim/action/RotateAbsolute_SendGoal
ServiceCaller.on_refresh_services_button_clicked(): could not get class turtlesim/action/RotateAbsolute_SendGoal of service /turtle1/rotate_absolute/_action/send_goal. Skipping it.
```

I am a ros2 beginner and I am following [this](https://index.ros.org/doc/ros2/Tutorials/Topics/Understanding-ROS2-Topics/#ros2topics) guide.
Thanks in advance

dirk-thomas, 2020-07-23:
> I am a ros2 beginner and I am following this guide.

The referenced tutorial doesn't mention invoking `rqt` anywhere - onlt `rqt_graph`.

What version of ROS 2 are you using? Did you build it from source or downloaded an archive with binaries? In either case which version / URL?

dirk-thomas, 2020-07-30:
@MicheleGalliano Friendly ping.

MicheleGalliano, 2020-07-30:
I found this issue using rqt_graph after the download of the latest ros2 version building it from source. This issue doesn't affect the correct results from the turtlesim tutorial

dirk-thomas, 2020-07-30:
The four error messages for `RosPluginProvider._parse_plugin_xml() plugin file` are clear. They are due to the fact the C++ plugins aren't supported on Windows atm (https://github.com/ros-visualization/qt_gui_core/blob/6cab60142436ef3ca712856e402f1bda9150578f/qt_gui_cpp/CMakeLists.txt#L11-L15) in combination with ros2/ros2#914

Regarding the `Malformed srv message_type` / `ServiceCaller.on_refresh_services_button_clicked(): could not get class` error message please provide reproducible steps how you ran into those.

MicheleGalliano, 2020-08-04:
Thank you, I don't remember what I modified since then but now I don't get that "malformed srv message_type" error anymore.

dirk-thomas, 2020-08-04:
> I don't remember what I modified since then but now I don't get that "malformed srv message_type" error anymore.

Ok, with the other problem fixes by the referenced PR I will go ahead and close this ticket. If you happen to run into the problem again please feel free to comment and the ticket can be reopened.

closed 2020-08-04 by dirk-thomas

Hello970, 2020-08-21:
Hi all, as a beginner of ROS2, I have met the same problem, just as shown in the top of this page. I would like to know is there an effective method to solve it? Thanks for your attention.

BJ-90, 2020-10-06:
> @Hello970 If you are referring to the `RosPluginProvider._parse_plugin_xml()` messages a new enough build including the referenced PR from `qt_gui_core` will make them go away. The patch is included in the version `1.1.1` of the package which will be available in the next Foxy and Rolling patch releases (or already in the latest nightly builds.
> 
> If you are referring to the `[ERROR] [...] [get_service_class]: Malformed srv message_type:` messages: there was no information how to reproduce them so nothing has changed.
> 
> In general please see the [ROS support guidelines](http://wiki.ros.org/Support) and in the future raise questions like this on [answers.ros.org](http://answers.ros.org/) instead.

Hi I am new learning ROS. How can I update these files?
Downloading zip file and replace them, or install these plugins using PowerShell?
I am having this error `RosPluginProvider._parse_plugin_xml()`  when use 'rqt' command

drewhamiltonasdf, 2020-10-09:
I am getting :
[ERROR] [1602206778.133982100] [get_service_class]: Malformed srv message_type: turtlesim/action/RotateAbsolute_SendGoal
and
 [ERROR] [1602206778.128320100] [get_service_class]: Malformed srv message_type: turtlesim/action/RotateAbsolute_GetResult

It seems to happen when I open a new Service Caller in RQT.

AJBuilder, 2021-01-05:
I am getting the same issue. The plugin error isn't causing any problems, but the Service Caller not being able to return all of Turtlesim's service's is problematic for learning ROS with out a physical platform...

I get:
`RosPluginProvider._parse_plugin_xml() plugin file "C:\dev\ros2_foxy\share\rqt_gui_cpp/plugin.xml" in package "rqt_gui_cpp" not found
RosPluginProvider._parse_plugin_xml() plugin file "C:\dev\ros2_foxy\share\rqt_gui_cpp/plugin.xml" in package "rqt_gui_cpp" not found
[ERROR] [1609812143.458272700] [get_service_class]: Malformed srv message_type: turtlesim/action/RotateAbsolute_GetResult
ServiceCaller.on_refresh_services_button_clicked(): could not get class turtlesim/action/RotateAbsolute_GetResult of service /turtle1/rotate_absolute/_action/get_result. Skipping it.
[ERROR] [1609812143.464531500] [get_service_class]: Malformed srv message_type: turtlesim/action/RotateAbsolute_SendGoal
ServiceCaller.on_refresh_services_button_clicked(): could not get class turtlesim/action/RotateAbsolute_SendGoal of service /turtle1/rotate_absolute/_action/send_goal. Skipping it.`

Only a few of turtlesim_node's services appear (/clear, /kill, /reset, /spawn). I am unsure if this is a turtlesim problem or an rqt problem. Any ideas?

I'm currently using a Windows install from binary packages.

felixf4xu, 2023-05-26:
> The four error messages for `RosPluginProvider._parse_plugin_xml() plugin file` are clear. They are due to the fact the C++ plugins aren't supported on Windows atm (https://github.com/ros-visualization/qt_gui_core/blob/6cab60142436ef3ca712856e402f1bda9150578f/qt_gui_cpp/CMakeLists.txt#L11-L15) 

Anybody knows why it's not supported in Windows? technical reason, or just not implemented yet?