ros2/ros2_documentation #825: Eloquent installation instructions use pip for flake8

christophebedard, 2020-07-18:
For full context, see this `ros2_tracing` issue: https://gitlab.com/micro-ROS/ros_tracing/ros2_tracing/-/issues/97

The short version is that installing `flake8` from `pip` (so the most recent version) on Ubuntu 18.04 for Eloquent leads to test failures, e.g.:

```
build/tracetools_test/test_results/tracetools_test/flake8.xunit.xml: 1 test, 0 errors, 1 failure, 0 skipped
- tracetools_test flake8.xunit.missing_result
  <<< failure message
    The test did not generate a result file:
    
    Traceback (most recent call last):
      File "/usr/local/lib/python3.6/dist-packages/flake8/plugins/manager.py", line 157, in load_plugin
        self._load()
      File "/usr/local/lib/python3.6/dist-packages/flake8/plugins/manager.py", line 134, in _load
        self._plugin = self.entry_point.load()
      File "/usr/local/lib/python3.6/dist-packages/importlib_metadata/__init__.py", line 107, in load
        return functools.reduce(getattr, attrs, module)
    AttributeError: module 'pycodestyle' has no attribute 'break_around_binary_operator'
    
    During handling of the above exception, another exception occurred:
    
    Traceback (most recent call last):
      File "/root/ros2_eloquent/install/ament_flake8/bin/ament_flake8", line 33, in <module>
        sys.exit(load_entry_point('ament-flake8', 'console_scripts', 'ament_flake8')())
      File "/root/ros2_eloquent/build/ament_flake8/ament_flake8/main.py", line 75, in main
        max_line_length=args.linelength)
      File "/root/ros2_eloquent/build/ament_flake8/ament_flake8/main.py", line 176, in generate_flake8_report
        style = get_flake8_style_guide(flake8_argv)
      File "/root/ros2_eloquent/build/ament_flake8/ament_flake8/main.py", line 140, in get_flake8_style_guide
        application.find_plugins(config_finder)
      File "/usr/local/lib/python3.6/dist-packages/flake8/main/application.py", line 159, in find_plugins
        self.check_plugins.load_plugins()
      File "/usr/local/lib/python3.6/dist-packages/flake8/plugins/manager.py", line 415, in load_plugins
        plugins = list(self.manager.map(load_plugin))
      File "/usr/local/lib/python3.6/dist-packages/flake8/plugins/manager.py", line 302, in map
        yield func(self.plugins[name], *args, **kwargs)
      File "/usr/local/lib/python3.6/dist-packages/flake8/plugins/manager.py", line 413, in load_plugin
        return plugin.load_plugin()
      File "/usr/local/lib/python3.6/dist-packages/flake8/plugins/manager.py", line 164, in load_plugin
        raise failed_to_load
    flake8.exceptions.FailedToLoadPlugin: Flake8 failed to load plugin "pycodestyle.break_around_binary_operator" due to module 'pycodestyle' has no attribute 'break_around_binary_operator'.
  >>>
```

I'm wondering why the [Eloquent installation instructions](https://index.ros.org/doc/ros2/Installation/Eloquent/Linux-Development-Setup/#install-development-tools-and-ros-tools) use `pip` for `flake8`, while:

* Eloquent PR jobs use `apt`, e.g. http://build.ros2.org/job/Epr__rclcpp__ubuntu_bionic_amd64/lastBuild/consoleFull#console-section-10
* [Foxy installations instructions](https://index.ros.org/doc/ros2/Installation/Foxy/Linux-Development-Setup/#install-development-tools-and-ros-tools) use `apt`

Perhaps the `flake8` version should be "pinned" by using `apt` instead of `pip`?

nuclearsandwich, 2020-08-13:
In general the ROS 2 team has decided to use linters from PyPI in ROS 2 CI in order to benefit from the latest improvements in those tools. This does create a split between what's run on ci.ros2.org and what's used in build.ros2.org and the package building pipeline.

Along with this decision the ROS team has opted not to backport changes solely to address linter issues.

These two decisions have led me to effectively disable and/or ignore linter test failures in Dashing and I know in the past I've suggested that the same be done for Eloquent since the maintenance requirements are higher than the benefit.

The Eloquent docs recommend the PyPI installation of flake8 and friends because at one point Eloquent was the development tip for ROS 2.

I think it would be reasonable to update the docs for Dashing and Eloquent to switch linter installation to apt.

christophebedard, 2020-08-16:
Thanks for the explanation!

I realized that pytest was causing problems as well. I'll open a PR with changes similar to what I did for ros2_tracing.

christophebedard, 2020-10-05:
Closing, see https://github.com/ros2/ros2_documentation/pull/849#issuecomment-703702715

closed 2020-10-05 by christophebedard