ros2/ros2_documentation #828: export/download tutorials as a pdf. feature request

elfelround, 2020-07-22:
selfish request, but the current layout can be cumbersome to travel

i personally always scrape documentations to pdf so i can either print them to highlight them or have them on my ipad where i can highlight them, study them etc

would be good if the whole of the documentation could be exported as a pdf keeping the order of all the fields since sitting in the computer and going link by link is cumbersome

CodeBeginnerCn, 2021-07-26:
I think it's a good idea to have a PDF version. 
I wonder if it is finished now ? 
What should I do to convert it? 
Is there any command line can do it? like 
```bash
$ make latex
$ pdflatex
```

Thanks！

clalancette, 2021-08-02:
In theory, you should be able to do:

```
python3 -m sphinx -M latexpdf source build -c .
```

To have the sphinx documentation generator output PDFs.  Unfortunately, I have yet to be able to make it work; I keep getting errors while Latex is running.  We'll leave this on the backburner to look at eventually, but if you find a solution in the meantime we'd be happy to review a PR.