ros2/ros2_documentation #830: ros2_documentation/source/Tutorials/Launch-files-migration-guide.rst is outdated

Thom747, 2020-07-24:
ros2_documentation/source/Tutorials/Launch-files-migration-guide.rst still refers to XML launch files for ROS 2, which were replaced with Python files.
Either this tutorial should be redone completely or scrapped to prevent confusion.

clalancette, 2020-07-24:
ROS 2 has support for Python, XML, and YAML launch files presently.  So I don't think that this tutorial is wrong in any particular way.

That being said, we didn't introduce XML launch files until Eloquent, I believe.  So if you are on Dashing, you may not have support for them.

I'm going to close this out, but feel free to keep commenting.

closed 2020-07-24 by clalancette

Thom747, 2020-07-24:
Ah, yeah, I ran Dashing until today, so I wasn't aware of this change. My bad!