ros2/ros2_documentation #832: misunderstandable on-boarding guide

dirk-thomas, 2020-07-28:
Based on user feedback the current on-boarding guide (https://index.ros.org/doc/ros2/Contributing/ROS-2-On-boarding-Guide/) can be misinterpreted to be used by any developer wanting to contribute to ROS 2. The instructions mention to create branches in the upstream repos rather than in fork. It would be good to clarify that these instructions are only meant for people at Open Robotics atm.

maryaB-osr, 2020-07-29:
Is there a reason this is on our docs site? Shouldn’t we keep it internal?

dirk-thomas, 2020-07-29:
I don't know. This content is coming from way back when all this was in the GitHub wiki. I don't have a preference where the information continues to be hosted. Internal sounds fine to me too.

clalancette, 2020-08-02:
I have to agree that this looks like internal documentation.  Not only that, a bunch of this is out-of-date (though still salvageable).  So I'd vote for removing the page from external documentation and putting it internal somewhere.

clalancette, 2021-03-25:
I'm closing this as we removed this page completely from the documentation.

closed 2021-03-25 by clalancette