ros2/ros2_documentation #842: Improve ROS 2 installation instructions

hoerenj, 2020-08-09:
The installation instructions [installing ROS2 on Windows](https://index.ros.org/doc/ros2/Windows-Install-Binary/) do not include the installation of development tools like colcon, etc. It would be very useful if the instructions included everything that is needed for ROS2 development (as in [Building ROS 2 on Windows](https://index.ros.org/doc/ros2/Windows-Development-Setup/)).

Maybe this applies to the instructions for other platforms as well (can only speak for Windows right now).

Blast545, 2020-09-02:
Hi there @hoerenj I was recently assigned to take a look on this, mind telling me if this is still a valid concern? At the moment, besides the link you provided to [Development setup on Windows](https://index.ros.org/doc/ros2/Installation/Eloquent/Windows-Development-Setup/) there is a link pointing to a colcon tutorial with instructions on [how to install and use colcon](https://index.ros.org/doc/ros2/Tutorials/Colcon-Tutorial/) to develop your own packages.

If the concern is that part of those tutorials should be specific to the main page, consider that the main installation page targets users that won't require additional development tools. 

If you think there is something else missing on those pages, please let me know and I'd be glad to take a look to sort this out. Thanks!

Blast545, 2020-10-01:
I'm closing this because IMO the comments that started this issue seem to be outdated. @hoerenj Feel free to open the issue again and let me know if you think there's still something missing from Windows guide so I can try solving it :)

closed 2020-10-01 by Blast545